AlbrechtL/welle.io #211: RTL_TCP Input: Add "set gain mode 1"

lars18th, 2018-02-18:
Hi,

As I commented previously (https://github.com/AlbrechtL/welle.io/issues/196#issuecomment-365919278), using the RTL_TCP input I can obtain better results when running Welle.io.

So, I isolated **one** difference when using it in front of the "QIRX SDR" tool (however, I still prefer the Welle.io):

- When using the RTL_TCP input in Welle.io, the first commands are: `set agc mode 0` and `set gain 0`. However, the other tool includes another command: `set gain mode 1`. So, when I first start the other tool and this command it's sent to the RTL_TCP server, then if I start the Welle.io **after** then I get the best results.

So, as a conclusion I suggest to send also (or add to the configuration) the option for setting this command.

I hope this helps!

AlbrechtL, 2018-02-18:
Now you can help us. It would be nice if you could check what "agc mode" and "gain mode" really mean?
What are the differences?
Which command does what?

lars18th, 2018-02-18:
> Now you can help us.

I'll try! I search for all of this info and I'll share here.
Stay on hold!

lars18th, 2018-02-23:
Hi @AlbrechtL ,

> Which command does what?

Here you can found the "official" RTL-TCP server code for review the different commands (no other documentation exists):
https://github.com/osmocom/rtl-sdr/blob/master/src/rtl_tcp.c#L307

> It would be nice if you could check what "agc mode" and "gain mode" really mean?

- **agc mode**: CMD:0x08. Function: `rtlsdr_set_agc_mode()`
-- Declaration: https://github.com/osmocom/rtl-sdr/blob/2be1612e604f950ba21ef6ff12488eab62e66ad5/include/rtl-sdr.h#L293
-- Description: Enable or disable the internal digital AGC of the RTL2832.
-- Code: `rtlsdr_demod_write_reg(dev, 0, 0x19, on ? 0x25 : 0x05, 1);`

- **gain mode**: CMD:0x03. Function: `rtlsdr_set_tuner_gain_mode()`
-- Declaration: https://github.com/osmocom/rtl-sdr/blob/2be1612e604f950ba21ef6ff12488eab62e66ad5/include/rtl-sdr.h#L253
-- Description: Set the gain mode (automatic/manual) for the device. Manual gain mode must be enabled for the gain setter function to work.
-- Code: call back to `set_gain_mode();` that it's specific for each tuner type.

> What are the differences?

The difference is that: The command "Set AGC Mode" enables the autogain in the **RTL2832 chip**. Instead of this the "Set Gain Mode" enables the autogain in the **TUNER chip**.

So, I suggest to enable too the autogain in the tuner. With my device this improves the signal a lot!

You agree then to change the current init order: from "0x08","0x04" to "0x03","0x08","0x04" ?

Regards!

AlbrechtL, 2018-02-26:
Thanks for the description. You are welcome to send a pull request with the change.

lars18th, 2018-02-27:
Hi @AlbrechtL ,

> You are welcome to send a pull request with the change.

Thank you for your offer!
However, I still need to complete a dev environment for compiling Welle.io _for Windows_. I feel isn't trivial.

Nevertheless, the enhancement is quite simple to code!

- In the current code the function for sending the command "0x03" already exists: _CRTL_TCP_Client::setGainMode()_
- But this function isn't used.
- And the function where the initial commands "0x08" & "0x04" are sended is _CRTL_TCP_Client::TCPConnectionWatchDogTimeout()_
- So, it's only required to add this code to https://github.com/AlbrechtL/welle.io/blob/master/src/input/CRTL_TCP_Client.cpp#L353

```
            qDebug() << "RTL_TCP_CLIENT:" << "Successful connected to server";
            connected	= true;
+
+           setGainMode(isHwAGC ? 1 : 0);
+
            setHwAgc(isHwAGC);

            if(isAGC)
```

Regards! :smile:

lars18th, 2018-02-27:
Hi @AlbrechtL ,

I create the PR using the GitHUB UI: https://github.com/AlbrechtL/welle.io/pull/220

I hope you agree with it!

lars18th, 2018-02-27:
Hi @AlbrechtL ,

Futhermore, you're already sending this command with the RTL_SDR driver:
http://github.com/AlbrechtL/welle.io/blob/master/src/input/CRTL_SDR.cpp#L143

So, in fact to be in synch (RTL_TCP vs RTL_SDR) you missed this command in the RTL_TCP input (and this is the reason for low signal using the RTL_TCP version).
Please, commit the PR. :smile: 
Regards!

AlbrechtL, 2018-03-02:
The changes are included in e5fcfce031fc21196d45fa414117bca78b06f71b.

You can now test it
[20180302_e5fcfce_Linux_welle-io-x86_64.AppImage](https://bintray.com/albrechtl/welle.io/download_file?file_path=20180302_e5fcfce_Linux_welle-io-x86_64.AppImage)
[20180302_e5fcfce_Windows_welle-io-setup.exe](https://bintray.com/albrechtl/welle.io/download_file?file_path=20180302_e5fcfce_Windows_welle-io-setup.exe)

closed 2018-03-03 by AlbrechtL

lars18th, 2018-03-05:
Hi @AlbrechtL ,

Great! It works! Thank you!

However, now I see the **TUNER** gain is not set. That is, you never send a message "0x0d" (_set tuner gain by index_: http://github.com/osmocom/rtl-sdr/blob/master/src/rtl_tcp.c#L356).

With my RTL-SDR device, the signal improves with "2" or "3".

You agree to set this value too?

AlbrechtL, 2018-03-05:
I use rtlsdr_set_tuner_gain() (0x04). What is the difference to set_gain_by_index() (0x0d)?

lars18th, 2018-03-13:
Hi @AlbrechtL ,

> I use rtlsdr_set_tuner_gain() (0x04). What is the difference to set_gain_by_index() (0x0d)?

Good question!
After review the code of "rtl_tcp" server I see this:
https://github.com/osmocom/rtl-sdr/blob/master/src/rtl_tcp.c#L250

So, the response is: When using command "rtlsdr_set_tuner_gain()" (0x04) you need to know the different values for the used TUNER. However, using the command "set_gain_by_index()" (0x0d) you can pass only "0", "1", "2", "3", etc. and the driver will use the right parameters for the detected tuner. So, if the server supports the cmd (0x0d) the code will be more simple and effective, as you don't need to know which tuner is running.

I hope this can help you.
Regards.

AlbrechtL, 2018-03-13:
Thanks for the clarification. But if we pass the correct values there shouldn't be any differences.

lars18th, 2018-03-14:
Hi @AlbrechtL ,

> Thanks for the clarification. But if we pass the correct values there shouldn't be any differences.

You're right. Now I see it!
http://github.com/AlbrechtL/welle.io/blob/master/src/input/CRTL_TCP_Client.cpp#L288

In any case, perhaps for the case of an 'unknown' tuner you like to use "set_gain_by_index()"

Thank you very much for your good job! :+1: