AlbrechtL/welle.io #217: ETI bitstream input: alternative method

lars18th, 2018-02-25:
Hi @AlbrechtL , @andimik , @mpbraendli , @basicmaster, et all

This is a continuation of #199... (Sorry to reopen this topic, but I have good news!)

As a proof of concept I created a demo for this. See the next two screenshots:

[image]
[image]

As you can imagine, I don't live in the UK. So I can't receive the BBC DAB broadcasts.

What you can see in this screenshots are:

* A realtime generation of IQ samples from the BBC DAB bitstream extracted from Astra 4.5E 12303H pid  1061.
* A realtime play of these IQ samples inside the Welle.io in my Windows machine.

How I can do that?
* I created a realtime RTL-TCP server that reads the samples from a FIFO. This is a simple mod of the tool `rtl_mus` and you can see the PR from here: http://github.com/simonyiszk/rtl_mus/pull/3
* Then I use one SAT streamer (in fact a SAT>IP server plus a software tool for streaming) to stream the transponder to the multicast address 239.1.1.10:5018 (this address/port is pure random).
* Inside one Linux machine I run the tools `socat`, `ts2na`, `na2ni` and `ODR-DabMod` to get the pid 1061 from the stream and generate _in realtime_ an IQ signal of this stream.
* Then I execute my custom `rtl_mus.py` configured for reading from a FIFO. This fifo file is filled with the output of the previous chain.
* I configured in my Windows machine a simple TCP redirector (with `tcptunnel-0.8.exe`) for listening in 'localhost' and port '1234' and connect to the Linux machine where is running the previous process.
* Finally I only need to execute the Welle.io in RTL-TCP mode, and violà!

What I like to demonstrate with this? **The ETI-NI format is a viable format for DAB bitstream interchange, between a demodulator and a decoder.**

Think now on this: A lot of computation performance it's wasted in generating the IQ samples and decoding them. Will be more efective a direct sharing of the ETI-NI data.

And what you can implement this in Welle.io?
* First, the demodulator part can be a simple process that generates the ETI-NI data SYNCHRONOUS. The output can be done in the sdtout, and we can transfer it to the network using the Socat tool.
* Second, the Welle.io can implement a new ETI_TCP input, that can reuse the skeleton of the RTL_TCP input. This new module will read from a TCP socket connected to the Socat tool. However, in this case the data it's in ETI-NI format instead of IQ samples.
* My suggestion for simplicity it's maintain the same RTL_TCP protocol, because other commands than the frequency selection don't care. And until we can develop a full ETI_TCP server, the Socat tool can do the same work. The only limitation is the fixed frequency.

Here the list of used tools:
* ETI-Tools: http://github.com/piratfm/eti-tools (ts2na & na2ni).
* ODR-DabMod: http://github.com/Opendigitalradio/ODR-DabMod ('next' branch).
* RTL-Mus: http://github.com/simonyiszk/rtl_mus (with my PR).
* TCPTunnel: http://github.com/vakuum/tcptunnel

I hope you will consider this enhancement. :smile: 
Regards!

lars18th, 2018-02-25:
I'd forgotten to say that if necessary I can share some capture of the MPEG-TS with the DAB bitstream encapsulated. In case you don't have access to the sat signal, then you can do the same replaying this capture stream over your network.

Please let me know if you need this capture.
Regards.

andimik, 2018-02-25:
This file can already been downloaded from http://tipok.org.ua/node/46

andimik, 2018-02-25:
And what are the good news? You described what I am doing for around 2 years. Also others in Germany did it and even built a DAB Monitor with it.

lars18th, 2018-02-26:
Hi @andimik ,

Great if you're working on "similar" tools! :smile: 

> This file can already been downloaded from http://tipok.org.ua/node/46

Thank you! But this file is in ETI-NI format. My offer is a capture of the MPEG-TS source.
I'll prepare one for sharing with all.

> And what are the good news? You described what I am doing for around 2 years

Your user case (to my understanding) is: "generate a RF DAB mux using hardware from an ETI-NI source". This it's right?

But, my user case it's quite different: "decouple the DAB demodulator from the DAB player". So, using _any_ DAB source (RF signal, satellite distribution, etc.) you will generate a ETI-NI bitstream. And this bitstream is then processed by a _software_ DAB player. So the "tuner" doesn't be directly attached to the DAB player.

The challenge in my scenario it's use the ETI-NI bitstream when the source can have errors.

andimik, 2018-02-26:
Please consider that satellite feeds are not for consumers, but for professionals like Radio stations or service providers for re-broadcasting. So please feel lucky that these feeds have been discovered and analyzed. 

There is no need to implement your wish, you can of course use this workaround with ODR

lars18th, 2018-02-26:
>> This file can already been downloaded from http://tipok.org.ua/node/46

>Thank you! But this file is in ETI-NI format. My offer is a capture of the MPEG-TS source.
>I'll prepare one for sharing with all.

Here it's a dump of 90 seconds in full MPEG-TS format with only pids 1061, 1062 & 1063:
http://www.sendspace.com/file/kv7ij3

Please, @andimik , can you share it in your page? I think other users will be interested in testing it.
You can also write the command line example to use it: ` nc -l 0.0.0.0 18081 < ./ts2na -p 1061 -s 12 | ./na2ni 0x0425.ts`

Thank you for your help! :smile:

andimik, 2018-02-26:
Which page?

lars18th, 2018-02-26:
Hi @andimik ,

> Which page?

This page: http://tipok.org.ua/node/46
In section "Usage examples".
This isn't your blog?

andimik, 2018-02-26:
Come on :-( 

I think he is called Sergej and he is from Ukraine. 

I fear you are just claiming without contributing.

lars18th, 2018-02-26:
> Come on :-(
> I think he is called Sergej and he is from Ukraine.
> I fear you are just claiming without contributing.

Sorry @andimik ! I'm doing the best I can. I'm just someone who likes this hobby, but I have a clear goal.

So far I've done the following:

1. Read the full specification.
2. Define an objective.
3. Prepare a demo.
4. Code a tool (my "RTL_Mus" patch for fake dongle).
5. Share my results.

What am I doing wrong?
I'm only looking for people who like to contribute in this direction: _decouple the DAB demodulator from the DAB decoder using ETI-NI as the interchange format_.
I'm so sorry if this is bothering you in any way. :cry:

lars18th, 2018-02-26:
Hi to all,

To simplify this "brute-force-workaround" to connect _any_ ETI source to _any_ DAB player, here an over simplification (all-in-one) executing this pipeline command:

- `socat UDP4-RECV:5018,bind=239.1.1.10,ip-add-membership=239.1.1.10:eth0,reuseaddr - | ./ts2na -p 1061 -s 12 | ./na2ni | ./odr-dabmod-next -f /dev/stdout -F u8 | nc -l -k -q -1 0.0.0.0 1234`

In this example the ETI source is an MPEG-TS multicast stream in udp://@239.1.1.10:5018 that encapsulates a DAB ETI-NA bitstream in the pid 1061. And the output it's a regular IQ sample server (2048000 Msps with 8bit unsigned ints) in the TCP port 1234.

And you can obtain similar results using other ETI sources. However, you need to have (at time) a **clean ETI-NI** to input the ODR-DabMod (I'm working on resolve it).

Please, check it! :smile:

andimik, 2018-02-26:
How do you tune to the frequency. Seems this is missing in your howto

lars18th, 2018-02-26:
Hi @andimik ,

> How do you tune to the frequency? Seems this is missing in your howto...

The "Frequency" it's totally _fake_ in the sense that the DAB _player_ can't change it. So, you can scan in _any_ frequency. Everytime, even if you try to change the frequency in the Player, you will get the _same_ DAB bitstream. This it's a current limitation of this _proof-of-concept_!

However, if you ask about how to "tune" in the DAB _demodulator_, in this case my example it's using an streaming from a Satellite link. If you need more info I can explain more.

In any case I'm working on improving this. If you found it interesting, then please comment it! :smile:

lars18th, 2018-02-26:
Hi,

Here more info about how to use this "hack" for getting the signal from a real RTL_TCP tuner (or any other SDR compatible receiver). You can generate the ETI-NI bitstream using the "eti-cmdline-*" tool. For example:

`./eti-cmdline-rtl_tcp -H 192.168.1.31 -I 1234 -C 5A -G 1 -Q  | ./odr-dabmod-next -f /dev/stdout -F u8 | nc -l -k -q -1 0.0.0.0 1234`

In this case the output it's the same as in https://github.com/AlbrechtL/welle.io/issues/217#issuecomment-368538927
But here the source it's a RTL_TCP server running in 192.168.1.31:1234.
Change it for adapt it to your environment!

**But remember:** Here you're comming from IQ samples and doing a DAB demodulation, plus another IQ modulation (plus the last IQ demodulation). So, a waste of computation! However, if your DAB source it's handled by a powerful device, then you can send over the network only the ETI-NI bitstream (2Mbps) to multiple clients (even in multicast). And in your client you do run the re-modulation and the final DAB Play (demodulation and decoding). At time, this only saves network bandwith, nothing more!

Note: FYI, I'm working in try to improve the ORD-DabMod to be more robust to errors.

AlbrechtL, 2018-03-03:
Great demo. Thanks for documenting!
Since there are no more comments I close this issue. But you are welcome to discuss it here more.

closed 2018-03-03 by AlbrechtL

lars18th, 2018-03-05:
> Great demo. Thanks for documenting!

You're welcome!

> Since there are no more comments I close this issue. But you are welcome to discuss it here more.

Perhaps you would like to prepare a WIKI entry referring to this topic.
I'll continue to working on this.

k-otixxx, 2018-07-08:
I have  success with Dablin player from Ubuntu to play ETI-NI from enigma2 set top box(from satellite feeds) with eti-tools installed.All satellite DAB feeds(12,5W,4,5E,7E,30E) work form me .
I want to use welle.io to test these streams.
@lars18th 
Could you post some examples how to start  realtime RTL-TCP server from set top box  with eti-tools installed.

lars18th, 2018-07-11:
Hi @k-otixxx ,

It's all commented here: https://github.com/AlbrechtL/welle.io/issues/217#issuecomment-368565576

Summarize:
- Replace you current Dablin command (`... | dablin.exe ...`) with `./odr-dabmod-next -f /dev/stdout -F u8 | nc -l -k -q -1 0.0.0.0 1234` (in Linux).
- Install (in Linux) the `RTL-mus` and patch it with my PR: https://github.com/simonyiszk/rtl_mus/pulls
- Then start the RTL-mus to the Netcat serving port (1234/tcp).
- At the end configure the Welle.io to use as RTL-TCP the RTL-mus.

The concept is quite simple: the ETI bitstream is converted (modulated) with the "ODR-DABmod" tool, and the output served as I/Q samples with a "fake" RTL-TCP server.

I hope it helps!

Request to @AlbrechtL : Please write a simple wiki with this info and share it! A lot of users request about it. But the issue is closed. Thank you!

andimik, 2018-07-11:
> I'm working on improving this.

> I'm working in try to improve the ORD-DabMod to be more robust to errors.

> I'll continue to working on this.

https://github.com/Opendigitalradio/ODR-DabMod/graphs/contributors

I cannot find your name there...

Where can we find your contributions? I told you some months ago that you're just claiming and this is still true.

k-otixxx, 2018-07-11:
@lars18th
Thanks a lot!
[url=https://postimg.cc/image/6yjpu7p23/][img]https://s33.postimg.cc/6yjpu7p23/IMG_20180711_235257.jpg[/img][/url]
[url=https://postimg.cc/image/87c6ty92j/][img]https://s33.postimg.cc/87c6ty92j/IMG_20180712_003207.jpg[/img][/url]

andimik, 2018-07-12:
BTW You can add pics here, no need to use external hosters

lars18th, 2018-07-12:
Hi @andimik ,

> https://github.com/Opendigitalradio/ODR-DabMod/graphs/contributors
> I cannot find your name there...
> Where can we find your contributions? I told you some months ago that you're just claiming and this is still true.

- I'm not a contributor of ODR-DabMod.
- My work with DAB radio open source projects is limited to some "ETI-tools" enhancements (https://github.com/lars18th/eti-tools), and a path for RTL-Mus (https://github.com/lars18th/rtl_mus); plus some simple patches time to time to different bugs/enhancements of other projects.
 - My work in this area is to push in move to support in current DAB players the ETI Input (instead of the I/Q samples), with the idea of decoupling the DAB tuner from the DAB player. 
- My contribution to this project (Welle.io) is the _description_ of how to use it for play ETI streams. All is published in this thread, and it's completly true.

I can't do more. As now I'm very busy with other projects... so _all_ DAB stuff is paused for me.
So, as I mentioned before... _I hope it helps!_ :smiley:

AlbrechtL, 2018-07-17:
@lars18th 

> Request to @AlbrechtL : Please write a simple wiki with this info and share it! A lot of users request about it. But the issue is closed. Thank you!

Feel free to write a wiki page at: https://github.com/albrechtl/welle.io/wiki