AlbrechtL/welle.io #223: Getting huge amount of [X-PAD len] in Terminal

dutchievj, 2018-03-08:
Using **welle.io_1.0_Linux_x86_64.AppImage** and before that **welle.io_1.0-rc2_Linux_x86_64.AppImage** on **Ubuntu 16.04** with **RTL_TCP.**

On some dab+ channels I get a constant slow flow of [X-PAD len] and then another [X-PAD len] and another etc.

But on some other DAB+ channels I don't get them at all.

Whats does [X-PAD len] mean?

basicmaster, 2018-03-08:
This means that the respective station sends its Dynamic Label and/or Slideshow violating the DAB specs (and wasting bandwidth) - which in this case actually requires to ignore that data. Here it is processed nevertheless and this message is an indicator for that.

Can you tell me on which multiplexes/stations this happens?

mpbraendli, 2018-03-08:
I observe the same on radio TOP TWO+, in the SMC_D02 ensemble on 7D in Zürich.
MSC recording here: http://www.mpb.li/pub/toptwo.msc

dutchievj, 2018-03-08:
@basicmaster 

It seems that all Dutch public (NPO) channels tricker this flood.

>  RadioController: Name of ensemble: "NPO             "
RadioController: Found station "NPO Radio 4     " ( 8204 )
RadioController: Found station "NPO FunX        " ( 8209 )
RadioController: Found station "NPO 3FM Alterna." ( 8210 )
RadioController: Found station "NPO Soul & Jazz " ( 8206 )
RadioController: Found station "NPO Radio 5     " ( 8205 )
RadioController: Found station "NPO SterrenNL   " ( 8212 )
RadioController: Found station "NPO 3FM         " ( 8203 )
RadioController: Found station "NPO Radio 2     " ( 8202 )
RadioController: Found station "NPO Radio 1     " ( 8201 )
RadioController: Found station "NPO 3FM KX Radio" ( 8214 )
RadioController: Found station "NPO Radio 4 Con." ( 8215 )
RadioController: Found station "NPO FunX Dance  " ( 8216 )
RadioController: Found station "NPO FunX Sl.Jamz" ( 8217 )

But the other Dutch non public (government sponsored) channels don't give [X-PAD len] at all.

basicmaster, 2018-03-08:
OK, this is interesting. Could you provide a short (5min) ETI dump of the mux? E.g. using `dab2eti` or `eti-cmdline`.

dutchievj, 2018-03-08:
No idea how to do that. I just have the AppImage and I am using that with an RTL dongle in another system.

mpbraendli, 2018-03-08:
I'm afraid I also can't dump an ETI, because I only have my B200 with me.

basicmaster, 2018-03-08:
@mpbraendli:
Your MSC dump is already enough.

@dutchievj:
I will contact a dutchman I know to hopefully get the ETI dump, to be able to contact NPO about that.

AlbrechtL, 2018-03-16:
Are they any updates?

basicmaster, 2018-03-16:
Just yesterday I received the dump of the NPO ensemble and was able to confirm that it is indeed affected.

To no longer "bug" normal users with the mentioned message, I disabled the message in loose mode (that is used by welle.io): https://github.com/Opendigitalradio/dablin/commit/8d50cea6e0aa7669e226ede75ba0493011d87458

@AlbrechtL:
Note that since a recent change you have to set the used MOT application type using the `SetMOTAppType()` method, after instantiation as well as after calling `Reset()`. I'm not sure whether you parse the User Application FIG: If so, you can use the parsed application type here. If not, use the hard coded default value `12`, which leads to the same behaviour as before the change.

AlbrechtL, 2018-03-18:
Thanks! I will update the X-PAD code soon.

AlbrechtL, 2018-08-30:
I updated the X-PAD decoder in https://github.com/AlbrechtL/welle.io/commit/86eb3509331d99d6b859a50e2385139fce4ee328.

closed 2018-08-30 by AlbrechtL

andimik, 2018-08-30:
Will the next branch also get this?

AlbrechtL, 2018-08-30:
It is in the next branch only.

andimik, 2018-08-30:
You are right, sorry.