AlbrechtL/welle.io #224: [Android] Welle.io using rtl_tcp and without the Martin Marinov's driver

lars18th, 2018-03-08:
Hi @AlbrechtL ,

This is a request based on the issue #187 .

My current objective is running Welle.io **with** Android Auto. And yes, I have all the requirements (AA HU, Android smartphone, RTL device, etc.).

As you know, to use AA your device needs to be connected to the car with the USB wire. So, you **can't** use at all a physical RTL dongle connected to it. So, at time I can start Welle.io in the Music section of the AA, but all the time the device driver is missing.

So, I started to work with a simple TCP redirection (listening over "0.0.0.0:1234" and redirecting the TCP to another device in the same wifi network running the RTL-TCP server). If you like to test it the App is this:
http://play.google.com/store/apps/details?id=com.elixsr.portforwarder

However, the problem is that the Android version of the Welle.io **insists** to open then Marinov's driver! Instead of trying to connect to the TCP socket first, and then if it fails, call to open the driver.

Here the code where the TCP connection is opened:
http://github.com/AlbrechtL/welle.io/blob/master/android/java/io/welle/welle/DabService.java#L1149

So, I suggest to modify this behaviour: First try to connect to the socket, and only if it fails, then call to open the driver.

You agree?
This will solve a lot of troubles using Android Auto.

Thank you!

lars18th, 2018-03-08:
Hi,

In fact, the code involved is this:

- Call to the driver: 
https://github.com/AlbrechtL/welle.io/blob/master/android/java/io/welle/welle/DabDelegate.java#L30
- Process Intent response: https://github.com/AlbrechtL/welle.io/blob/master/android/java/io/welle/welle/DabDelegate.java#L122

The solution, can be at the start of `DabDelegate::startSdrActivity()` (I write with C++ syntax, not Java) create the Intent with action `ACTION_SDR_DEVICE_ATTACHED` and send it to the `DabService`, like to code from here: https://github.com/AlbrechtL/welle.io/blob/master/android/java/io/welle/welle/DabDelegate.java#L127
 This will connect to the socket, even if the driver is not running or not exists. And only in case of fail, retry with the driver.

OR if you prefer, use a configuration variable for use directly the TCP socket or the DRIVER, but just only one. Personally, I prefer this last option. And in this case, you can save also the IP:PORT of the RTL-TCP server.

What you think?

lars18th, 2018-03-14:
Hi @AlbrechtL ,

You know how to resolve this issue?

AlbrechtL, 2018-03-16:
In my point of view that best would be to add an option to set the input driver and in case of rtl-tcp the IP address and IP port. You are welcome to send a pull request.

lars18th, 2018-03-16:
Hi @AlbrechtL ,

> In my point of view that best would be to add an option to set the input driver and in case of rtl-tcp the IP address and IP port. 

Me too!
However, the current code doesn't work if you simple change the IP address and IP port. The problem is that the current implementation depends on the Marinov's driver. It can't start the TCP socket if the driver isn't detected+activated+running.
So, for this reason I request the modification.

> You are welcome to send a pull request.

And I'll try regarding the UI. However, the current behaviour needs to be modified. Instead of first call to the driver, is best to simple test the TCP socket connection **before**, and **only** if it fails then use the current method of call to the driver.
The problem is: I'm not a programmer of Java and Android. I only have skills of C and Linux. So this part (the Java sections) is out-of-scope for me.

You agree to update this part?

rubenverhoef, 2018-04-21:
So i cannot use android welle to connect to my RTL_SDR server on my RPI right now?
Is this function going to be implemented in the next version?

I want to use welle with android auto and i have to use the RTL_SDR for it. Now the Android Auto support is only for the android auto version on the phone and not on my headunit, right?

lars18th, 2018-04-24:
> So i cannot use android welle to connect to my RTL_SDR server on my RPI right now?
Is this function going to be implemented in the next version?

I hope it!

> want to use welle with android auto and i have to use the RTL_SDR for it. Now the Android Auto support is only for the android auto version on the phone and not on my headunit, right?

Android Auto run *all the time* in the smartphone. The HeadUnit is only a *remote screen*. More or less like your Monitor regarding your Desktop Computer (the monitor is dumb, and the PC is smart... with AA the _monitor_ is the HeadUnit, and the _PC_ is the SmartPhone).

rubenverhoef, 2018-04-24:
True, but we cannot use USB dab receiver at the same time as the Android Auto USB connection. Hope this will made possible soon :)

lars18th, 2018-04-24:
> True, but we cannot use USB dab receiver at the same time as the Android Auto USB connection. Hope this will made possible soon :)

Sure! This is the reason for supporting plain RTL-TCP connections (that is, user configuration of the target IP:PORT) in the Android version. Then you can receive the sample stream over the WIFI connection of the Smartphone from an standalone server (Pi or another Android device, for example). This other device runs in the car with the USB dongle and the antena, with the role of the RTL-TCP server.

The problem now is that the use of the Martin Marinov's driver is *hardcoded* in the Andorid version, even if it uses a simple TCP connection with the driver. So, without running this driver, the Welle.io refuses to run.

I hope soon this will be improved.

rubenverhoef, 2018-05-06:
@lars18th One question, how are you going to send the RTL_SDR to your phone?
I have a raspberry pi which sends a hotspot without internet, when i connect my android phone the rtl-sdr server is not accessible because the hotspot has no internet connection and my phone is looking for the rtl-sdr server on my mobile network. I can select an option on my phone to use the wifi network but then i have no internet because it is using only wifi (without internet access).

So we need to have a hotspot with internet connection or do you have an other option?

lars18th, 2018-05-07:
Hi @rubenverhoef ,

This is my deployment:

- I have a car with AA support (and WiFi, but this isn't relevant for this user case).
- I installed an App in the phone that when the phone is connected to the Bluetooth car, then it activates the WiFi hotspot in my phone. With this network, any device inside the car can connect to the Internet, and also to the phone.
- I like to install a Pi3+ in my car, with a RTL-SDR and a DAB antenna. It will connect, using the internal WiFi support to the hotspot network of my smartphone. At time, I do the same using my laptop for testing. All works (so, it's connectivity between the smartphone and the RTL device).
- And last, I need that the Welle.io in the phone will connect to the RTL-SDR server in the Pi3+ (or similar device connected by the WiFi network). However, the problem is the current hardcoded use of the Martin Marinov's driver.

Remember: AA uses USB+Bluetooth, so you have free the WiFi connection. And you can use it as a hotspot (wifi tetherring).
So, I hope soon this restriction will be removed.
Regards.

rubenverhoef, 2018-05-07:
Yeah of course, an hotspot from the phone. That will work, but i personally think an hotspot from the RPI would be better because my RPI doesn't need an internet connection and it is more user friendly in my eyes.

lars18th, 2018-05-07:
Hi @rubenverhoef ,

Your proposal has one relevant drawback:

- If the Pi3 (that's the RTL server) runs as WiFi Hot-Spot, then the phone try to use it as a gateway, and this is a problem.

But, if the Pi3 is a client, then you can disable the gateway (Internet access) in it to not use the smarphone's Internet connection. Or you can create firewall rules to block it. In any case, it's more easy to manage the RTL sever than the Android device. So, the WiFi tethering is the best solution.

Regards.

lars18th, 2018-09-20:
Hi @AlbrechtL ,

Any news about manual configuration for the Android version?

AlbrechtL, 2018-09-21:
Yes. I have planned to make the rtl_tcp input configurable via the GUI. This will be also available for Android. But the Android build is broken at the moment (https://github.com/AlbrechtL/welle.io/issues/287). So there is nothing to test.

lars18th, 2018-09-25:
> Yes. I have planned to make the rtl_tcp input configurable via the GUI. This will be also available for Android. But the Android build is broken at the moment (#287). So there is nothing to test.

Great to hear it, @AlbrechtL !

So, when the Qt 5.12 will be available you'll try to re-enable the Android build? And after you will implement the configurable rtl_tcp input from the GUI? Nice!

More or less the ETA will then be around the new year. So, we need to wait patiently.
No problem for me. Great job! :+1:

AlbrechtL, 2018-11-16:
rtl_tcp in now configurable (https://github.com/AlbrechtL/welle.io/commit/5e67dcd046a9d2eb276d4259270465575c1e17bc) and the Android build is working again with Qt 5.12. 

BTW: For early bird testers: Martin Marinov's driver is not working at the moment.

closed 2018-11-16 by AlbrechtL

rubenverhoef, 2018-11-16:
Any change i can find/you can give me an (ARM)APK or do i need to compile it myself?

lars18th, 2018-11-16:
Hi @AlbrechtL ,

> rtl_tcp in now configurable (5e67dcd) and the Android build is working again with Qt 5.12.

Thank you! I'll test it.

> BTW: For early bird testers: Martin Marinov's driver is not working at the moment.

We can't test it at all, as the nightly builds doesn't have any Android version (the last Release is the version 1.0). Futhermore, the Windows version in the nightly is previous to this https://github.com/AlbrechtL/welle.io/commit/5e67dcd046a9d2eb276d4259270465575c1e17bc.
https://bintray.com/albrechtl/welle.io/welle.io_nightly#files

So, please, can you post a "beta" for all platforms to test it (like an RC)?
And remember, pleae, to re-open the issue; so we can provide some feedback.

Thank you!

AlbrechtL, 2018-11-18:
Once Martin Marinov's driver is working again I will provide an APK. 

>  And remember, pleae, to re-open the issue; so we can provide some feedback.

You can give feedback in this issue.

pathfinder1803, 2021-06-06:
Hi,
the rtl_tcp function isn't included yet, is it? I can't find a option to select. Shall I start a new issue as this one is closed?
Thanks in advanced

lars18th, 2021-06-07:
> the rtl_tcp function isn't included yet, is it? I can't find a option to select. Shall I start a new issue as this one is closed?
> Thanks in advanced

Hi,

The README continues to say that the Martin Marinov's driver is **mandatory**: https://github.com/AlbrechtL/welle.io#android
IMHO I can't see any valid reason to not support a remote RTL_SDR server (or any other TCP driver) in Android devices.

pathfinder1803, 2021-06-07:
> Hi,
> 
> The README continues to say that the Martin Marinov's driver is **mandatory**: https://github.com/AlbrechtL/welle.io#android
> IMHO I can't see any valid reason to not support a remote RTL_SDR server (or any other TCP driver) in Android devices.

The problem is, that only one USB OTG device seems to be supported. If the RTL SDR Dongle is plugged in, I can't connect to Android Auto anymore.
The idea is to use TCP for the signal and the USB for the car. That's the valid reason. And I'm pretty sure there will be more people with this request as the end of FM is close.

lars18th, 2021-06-07:
Hi @pathfinder1803 ,

> The problem is, that only one USB OTG device seems to be supported. If the RTL SDR Dongle is plugged in, I can't connect to Android Auto anymore.
> The idea is to use TCP for the signal and the USB [...]

Yes, the problem is that the **current** implementation enforces to use the Martin Marinov's driver. And this implies the use of a USB device without any alternative. And for this reason **three** years ago I've opened this issue requesting to support TCP sources in the Android App. But @AlbrechtL has closed the issue when he has added a commit with some support: https://github.com/AlbrechtL/welle.io/issues/224#issuecomment-439302942

However, I feel the problem already exists. So, please request to him about this. IHMO it has no sense to support only USB devices when the Android Auto connection requires to use the USB port.