AlbrechtL/welle.io #232: Get RTL_TCP to work again.

mpbraendli, 2018-03-25:
We have lost the RTL_TCP input, because there is no cross-platform replacement for the QT sockets. We don't want to add a dependency to boost.
Option 1: Use socket API (different on Win32 and POSIX)

mpbraendli, 2018-03-26:
POSIX socket interface added in e8ff90f
Win32 support still needed! I can't do this myself, I have no way to test!

gvanem, 2018-03-26:
What? I use Windows-10 and RTL_TCP input works fine. I use this `well-io_rtl_tcp.bat` to start it:
```
@echo off
setlocal
cdd f:\gv\dx-radio\Welle-IO\src\bin
rm -f welle-io_rtl_tcp.log
start f:\gv\dx-radio\Osmocom-SDR\src\rtl_tcp.exe -d 1 -p 1235 -f 225560000
welle-io.exe --device rtl_tcp --rtl_tcp-address 127.0.0.1 --rtl_tcp-port 1235 --log-file welle-io_rtl_tcp.log
```

mpbraendli, 2018-03-26:
This is for the `next` branch.

lars18th, 2018-03-28:
Hi @mpbraendli ,

Great idea! :smile: 
If you like to support _multiplatform_ (Windows/UNIX) sockets with much of code and without adding library dependencies, then you can try `Knot`: http://github.com/r-lyeh-archived/knot
You only need to add two files: `knot.cpp` and `knot.hpp`, and use the simple C++ API.

Regards!

mpbraendli, 2018-03-28:
Looks like a cute little library, thanks for the hint.

basicmaster, 2018-03-28:
>  Option 1: Use socket API (different on Win32 and POSIX)

This should still be an option (instead of adding a further dependency), as the POSIX socket API is also supported on Windows; I used it some years ago for a multi platform project. See https://msdn.microsoft.com/en-us/library/ms740096(v=vs.85).aspx

Two of the differences that I remember were that you have to init WinSock and that also some constants had other names.

AlbrechtL, 2018-04-05:
Added in 83303e5692532e5909de16cc2026b18ae91da68d.

closed 2018-04-05 by AlbrechtL