AlbrechtL/welle.io #237: PlutoSDR not recognized

beta-tester, 2018-04-06:
i have a Analog Devices ADALM-Pluto SDR (rev. B), the range extension hack is applied (60MHz-6000MHz) and all its IIO driver are installed (Windows 10).
but welle.io v1.0 does not see the PlutoSDR device at all.
the example application IIO - oscilloscope from AD is working fine...
[ADALM-PLUTO for End Users](https://wiki.analog.com/university/tools/pluto/users)

is there a way to get PlutoSDR work with welle.io application?

AlbrechtL, 2018-04-07:
At the moment there is no direct way.

I you compile welle.io from the sources for Linux you might can use the SoapySDR input with the [PlutoSDR SoapySDR Plugin](https://github.com/jocover/SoapyPlutoSDR).

closed 2018-05-23 by AlbrechtL

beta-tester, 2018-05-24:
is it not possible to pick some code from here [(PlutoSDR input plugin)](https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/plutosdrinput)

mpbraendli, 2018-05-24:
It's always possible to pick code somewhere, but somebody that owns a PlutoSDR should do it and maintain it. I would prefer seeing the same effort invested in getting the PlutoSDR to work with SoapySDR, because more projects can make use of it.

beta-tester, 2018-05-24:
oops... i didn't realized that [SoapySDR](https://github.com/pothosware/SoapySDR/wiki) is "A fresh and clean vendor neutral and platform independent SDR support library"

but is [PlutoSDR SoapySDR](https://github.com/jocover/SoapyPlutoSDR) also available for Windows OS?

mpbraendli, 2018-05-24:
It's better to ask them (whoever "they" are), I don't know.

AlbrechtL, 2018-05-27:
@Waitsnake compiled SoapySDR for Windows. See here https://github.com/AlbrechtL/welle.io/pull/253. But I don't know the current state.

mpbraendli, 2018-05-28:
It's not SoapySDR itself that I'm worrying about, it's the PlutoSDR device driver for SoapySDR.

beta-tester, 2018-07-05:
i have an additional stupid question.
i just installed PothosSDR with SaopySDR for windows
(see [SoapySdR please include drivers for plutosdr --windows 10](https://github.com/pothosware/PothosSDR/issues/53) )
my hope was, that i can manage it to get welle.io to work with this...
but i failed... :(
any idea what i have to do without compiling anything?

AlbrechtL, 2018-07-06:
At the moment there is no way to do it without compiling.