AlbrechtL/welle.io #238: show "ppm" in addition to frequencey correction

beta-tester, 2018-04-06:
there is frequency correction shown on the expert view.
is it possible to show the ppm value as well?

10^6 * -freq_correction / freq_tuned = ppm

e.g.:
current channel: 5C (178.352 MHz)
freq. correction: -11879 Hz (66.60 ppm)

 think this is an easy to implement feature, but would help to calibrate the SDR device for other applications.

beta-tester, 2018-04-27:
i have implemented that ppm addition...
here the patch file of the changes:
[patch-ppm.txt](https://github.com/AlbrechtL/welle.io/files/1954048/patch-ppm.txt)

beta-tester, 2018-04-27:
in **ExpertView.qml** maybe change
`cppRadioController.FrequencyCorrectionPpm`
to
`cppRadioController.FrequencyCorrectionPpm.toFixed(2)`

to have only 2 digits after the point.

AlbrechtL, 2018-04-28:
Thanks for the idea.

It would be nice if you could send a pull request for the [next branch](https://github.com/AlbrechtL/welle.io/tree/next).

beta-tester, 2018-04-30:
ppm is merged to the next branch.
so, hopefully it can be seen in the next release...

closed 2018-04-30 by beta-tester

AlbrechtL, 2018-04-30:
Please review my comment: https://github.com/AlbrechtL/welle.io/pull/243#issuecomment-385421261