AlbrechtL/welle.io #239: Provide an Audiostream/API

martenumberto, 2018-04-19:
Is it possible to provide an Audiostream and/or an API?
I'm trying to build an app for Android Auto to play music over the OpenAuto project.
The decoding should be done by an Raspberry (with attached RTL-SDR Stick) only the Playback, Control and the Information should be done by the Android app

mpbraendli, 2018-04-19:
I'm working on an API in welle-cli, with the primary purpose to do some monitoring. See `next` branch, compile with CMake. Work in progress!

AlbrechtL, 2018-08-18:
Are they any updates to this topic?

mpbraendli, 2018-08-19:
I would argue that `welle-cli` now offers an HTTP API that would be usable for such an application. As before: work in progress.

mpbraendli, 2018-08-28:
Closing the ticket because I believe the desired application is doable with welle-cli.

closed 2018-08-28 by mpbraendli