AlbrechtL/welle.io #241: Error on 64bit linux: QtCharts not installed

ahahn94, 2018-04-24:
When building welle.io on Solus linux (which is 64bit only) using cmake, I get an error when trying to start the program:
```
main: Set language "de_DE"
Audio: ActiveState
QQmlApplicationEngine failed to load component
qrc:/src/gui/QML/main.qml:624 Type ExpertView unavailable
qrc:/src/gui/QML/ExpertView.qml:72 Type SpectrumView unavailable
qrc:/src/gui/QML/SpectrumView.qml:2 module "QtCharts" is not installed
```
Qt5 Charts as well as the developement files for it are installed. I have libQt5Charts.so etc. in my libraries path. 

Does welle.io require Qt4 for the charts, or does it require the 32bit version?

Greetings

AlbrechtL, 2018-04-25:
64 bit is supported as well as QT 5. The charts module comes with QT 5.6 or higher. Please make sure that you have a consistent QT installation.

ahahn94, 2018-04-26:
The whole Qt5 Stack of Solus is version 5.10.0, so the version is consistent between the modules.
I also tried running some of the QtCharts demos from the qt5-charts-demo package of Solus. No problem there, the modules works as intended. But welle.io does not find the module...

AlbrechtL, 2018-04-26:
Can you try to compile welle.io via qmake or QT Creator?

ahahn94, 2018-04-26:
I tried building via qmake as follows:
```
cd build
sudo rm -r *
qmake ..
make
```

I had to disable airspy for the build as there is no package for it on solus.

Sadly, the problem persists with the new binary...

AlbrechtL, 2018-04-26:
At the moment I have no idea. Can you verify if the Solus QTCharts package includes the QML part?