AlbrechtL/welle.io #242: Project ERROR: Unknown module(s) in QT: charts

beta-tester, 2018-04-26:
i have Ubuntu 16.04.x LTS and i tried to follow the build description [Ubuntu Linux 16.04 LTS](https://github.com/AlbrechtL/welle.io#ubuntu-linux-1604-lts)

i downloaded "Open Source Usage under (L)GPL v3 license" -> "Go open source" -> `qt-unified-linux-x64-3.0.4-online.run`

i installed QT 5.9 including the QT Charts 
[image]

i installed the packages as shown
`sudo apt install libfaad-dev libfftw3-dev librtlsdr-dev libusb-1.0-0-dev mesa-common-dev libglu1-mesa-dev libpulse-dev`

i cloned the project `git clone https://github.com/AlbrechtL/welle.io.git`

but when i open the file `~/welle.io/welle.io.pro`
i get the following error in Qt Creator.
```
Project MESSAGE: Current git hash = 4e13f5e
Project ERROR: Unknown module(s) in QT: charts
Warnings while parsing QML type information of /usr/lib/x86_64-linux-gnu/qt5/qml/Ubuntu/Content:
Failed to parse "/usr/lib/x86_64-linux-gnu/qt5/qml/Ubuntu/Content/plugins.qmltypes".
Error: /usr/lib/x86_64-linux-gnu/qt5/qml/Ubuntu/Content/plugins.qmltypes:276:19: Expected string literal to contain 'Package/Name major.minor' or 'Name major.minor'.
/usr/lib/x86_64-linux-gnu/qt5/qml/Ubuntu/Content/plugins.qmltypes:277:36: Expected array literal with only number literal members.
/usr/lib/x86_64-linux-gnu/qt5/qml/Ubuntu/Content/plugins.qmltypes:283:19: Expected string literal to contain 'Package/Name major.minor' or 'Name major.minor'.
/usr/lib/x86_64-linux-gnu/qt5/qml/Ubuntu/Content/plugins.qmltypes:284:36: Expected array literal with only number literal members.
/usr/lib/x86_64-linux-gnu/qt5/qml/Ubuntu/Content/plugins.qmltypes:301:19: Expected string literal to contain 'Package/Name major.minor' or 'Name major.minor'.
/usr/lib/x86_64-linux-gnu/qt5/qml/Ubuntu/Content/plugins.qmltypes:302:36: Expected array literal with only number literal members.
```
i tried Qt 5.10 instead of Qt5.9...
i also tried to install Qt as root...
... all without any difference and no success to build/run welle.io.

andimik, 2018-04-26:
Same here. It worked for me last time I built it, I guess mid of March.

beta-tester, 2018-04-27:
i deleted the Qt folders
```
rm -rf ~/Qt/
rm -rf ~/.config/Qt/
rm -rf ~/.config/QtProject/
```

then i did really a full installation of Qt 5.9.5 with really **ALL** its sub-items (3.19 GByte).
(i think i could skip "Sources", but i wanted to be really sure)
[image]
now it works... i can build and debug/run the welle.io application.
(last time where i said, that i marked all sub-items, i skipped Sources and Android items - oops)

thank you for your help and the really nice piece of software "welle.io"

beta-tester, 2018-04-27:
ok, this works as well...
takes less disk space
```
rm -rf ~/Qt/
rm -rf ~/.config/Qt/
rm -rf ~/.config/QtProject/
rm ~/.config/QtProject.conf
```
and
[image]

andimik, 2018-04-27:
BTW: Is it really necessary to install the complete Qt?

On a fresh Kubuntu I get with `cmake`

>   Could not find a package configuration file provided by "Qt5Quick" with any
>   of the following names:
> 
>     Qt5QuickConfig.cmake
>     qt5quick-config.cmake

But with `qmake` I get some more missing packages

> Project MESSAGE: Current git hash = da73c76
> Project ERROR: Unknown module(s) in QT: qml quick charts multimedia

AlbrechtL, 2018-04-28:
At the moment cmake is not supported with a stock Ubuntu and it derivatives.

It close it because this issue is solved.

closed 2018-04-28 by AlbrechtL

AlbrechtL, 2018-04-28:
I just tried Ubuntu 18.04 LTS with cmake successfully.

Please install the following packages
```
qtbase5-dev
qtdeclarative5-dev
qtmultimedia5-dev
libqt5charts5-dev
qml-module-qtcharts
qml-module-qtquick-controls2
qml-module-qtquick-controls
qml-module-qt-labs-settings
```

andimik, 2018-05-05:
No, there is an error:
> qrc:/src/gui/QML/MainView.qml:43 module "QtQuick.Controls" version 2.3 is not installed

(I'm trying to build the next branch)

AlbrechtL, 2018-05-10:
The next branch requires QT 5.10 which is not shipped with Ubuntu 18.04. Ubuntu 18.04 only includes QT 5.9.
To build the next branch you have to use the QT Online Installer to install QT 5.10.

dirkhohmann, 2019-05-11:
O.k. folks, I tried to build on Ubuntu 18.04 in May 2019. There's no QT 10 5.10 available. I tried QT 9.13, 9.5 always missing "QT charts" - I don't wanna waste my time on this broken stuff - good bye

AlbrechtL, 2019-05-23:
Please follow the https://github.com/AlbrechtL/welle.io/blob/next/README.md and install Qt 5.10 or above. Then you will be able to build welle.io.

dirkhohmann, 2019-05-23:
I follwed the instructions in the readme, simple straight forward but reality was different.  After several attempts with different Qt configurations I could successfuly build welle.io.
It seems that the dependencies are not really clean.

AlbrechtL, 2019-05-30:
Which dependencies do you mean?