AlbrechtL/welle.io #256: [next] Investigate lock loss even with strong signals in multipath

mpbraendli, 2018-05-07:
The receiver behaves badly in presence of two signal components (two peaks in CIR) when the first peak is weaker than the second. The receiver locks onto the stronger one, but should lock on the earlier one instead. Locking on the second leads to increased inter-symbol interference, and can kill reception.

Maybe related to issue 247

mpbraendli, 2018-05-31:
Some progress in `next` branch. https://twitter.com/mpbraendli/status/1002088951488962561

AlbrechtL, 2018-07-19:
Is there is any progress here?

mpbraendli, 2018-07-26:
When using good receivers, the sync time can be improved by disabling the coarse frequency correction (now in the settings page). With the new sync mechanism, behaviour in multipath scenarios is much improved, which is now default, but can be turned off in the settings page.

closed 2018-07-26 by mpbraendli

AlbrechtL, 2018-08-03:
Great!

Do you have a rtl-sdr dongle that fulfills the good receiver requirement?

mpbraendli, 2018-08-06:
I do have two rtlsdr dongles that have a frequency offset that is less than 100Hz, which is more or less what we could define as a criteria for "good". On those, the receiver is quicker to get lock with coarse correction disabled. I probably have an old "bad" one in a box somewhere, but haven't been using it for a while.

The USRP B200 and the LimeSDR mini have accurate clocks too.

AlbrechtL, 2018-08-06:
Now it becomes complicated how we can document this on the website. Are using your good rtl-sdr a TCXO?

mpbraendli, 2018-08-07:
Well, I'd write something along the lines of "if you observe that the frequency correction is smaller than 300Hz with your device, you can disable the coarse frequency corrector, as it isn't needed and can make channel switching faster".

I have one with about 400Hz offset, it also works well without coarse corrector.

My "good" rtlsdr happens to have a TCXO, yes.

EPO35F, 2019-04-05:
Hello, I have installed to receivers, start 2 welle-io.exe, and only the first one works.
Any hints how to have several ones working ?
I am mainly interested by the data within DAB
I have Nooelec receiver and RTLSDR RTL2832U V3 ones

AlbrechtL, 2019-04-05:
Please open an new issue for your request or search on the old issues because we already had this question.