AlbrechtL/welle.io #260: Adding support for Airspy (Mini) Bias Tee

Kabuse, 2018-05-15:
I am using an Airspy Mini and an outdoor antenna with a preamplifier mounted directely to the antenna. But since there is no option available in welle.io to turn the internal Bias Tee of the Airspy (Mini) on and off, I can not use welle.io with my setup. So, an option (checkbox) for enabling the Bias Tee, when using an Airspy (Mini) should be integrated.

AlbrechtL, 2018-07-24:
I plan to make the receiver dongles configurable via the GUI. Then we can add this as well (honestly this is more a low priority task).

AfromD, 2018-09-10:
Same request for another great dongle; the RTL-SDR also has a software-activated bias tee.
https://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/

AlbrechtL, 2018-11-03:
In 9dfdf7fbb5a71941a8f4c3aba2b0a36027b09d15 I added rtl-sdr and Airspy bias tee support. Please test it and reports issues!