AlbrechtL/welle.io #261: [next] subch X bitrate:terminate called after throwing an instance of 'std::runtime_error'

r0k, 2018-05-27:
It seems that `Subchannel::bitrate()` fails on "DR Deutschland" at least on "Schwarzwaldradio" and "SCHLAGERPARADIES" which both claims to have EEP-B Level 5. The current implementation of `Subchannel::bitrate()` in backends  `dab-constants.cpp` goes up to level 3.

```sh
[subch 20 bitrate:terminate called after throwing an instance of 'std::runtime_error'
  what():  Unsupported protection (protOption: 1, protLevel: 5)
```
I first changed the logging to see why it fails and then i "added" the two sublevels by extending the number sequecene. There is for sure a more sophisticated approach possible but i didnt found the right information via bing :)

```cpp
        ...
        else { // EEP-B
            if (protLevel == 0)
                return length / 27 * 32;
            if (protLevel == 1)
                return length / 21 * 32;
            if (protLevel == 2)
                return length / 18 * 32;
            if (protLevel == 3)
                return length / 15 * 32;

            if (protLevel == 4)
                return length / 12 * 32;
            if (protLevel == 5)
                return length / 7 * 32;
        }
    }

    throw std::runtime_error("Unsupported protection (protOption: " + std::to_string(protOption) + ", protLevel: " + std::to_string(protLevel) + ")" );
```  

Hope that helps. Thanks for the cool work!

andimik, 2018-05-27:
But there is only 1B, 2B, 3B and 4B. There is no Protection Level 5 or even 6 for DAB+

r0k, 2018-05-27:
@andimik 
> But there is only 1B, 2B, 3B and 4B. There is no Protection Level 5 or even 6 for DAB+

All I can say is that `welle-cli` crashes on scanning channel `5C` at my location and this is the reason. Other channels work as expected.

This is why I extended the **throw std::runtime_error** to see what happened:

```
Unsupported protection (protOption: 1, protLevel: 5)
```
`protOption: 1` leads into the second else branch (EEP-B) and `protLevel: 5` is not covered by the current implementation (protLevel 0 to 3). 

Since this is coming in for two different stations it might be worth to look at it.

I re-searched a bit about the protection levels and the code covers the defined variants and levels. But still there is that `protLevel: 5` value coming in for the two stations. 

> For 'old fashioned DAB' there are 5 error protection levels, UEP 1 to 5. 
... 
In DAB+ this has changed, and modern Reed-Solomon encoding is used, protecting all audio bits the same way. There are 8 protection levels, in 2 categories A and B. EEP (Equal Error Protection) A1 to A4 have code rates from 1/4th, 3/8th, 1/2nd and 3/4th, and EEP B1 to B4 have code rates 4/9th, 4/7th, 4/6th and 4/5th . The EEP A protection can be used for bitrates in 8 kbits/sec increments, whereas the EEP B protection levels are limited to bitrates in 32 kbits/sec increments. 

This is exactly what the `Subchannel::bitrate()` code is about. Strange.

mpbraendli, 2018-05-28:
An IQ file recording of this ensemble would be useful for debugging, can you capture one?

According to ETSI EN 300 401, Clause 11.3.2, there is EEP with levels 1, 2, 3 and 4. These might be mapped to 0,1,2 and 3 respectively in this code.
But EEP level 5 doesn't exist. Probably a decoding problem.

andimik, 2018-05-28:
You mean 5C? I've uploaded a dump (TS) at https://github.com/piratfm/eti-tools/issues/12
https://github.com/piratfm/eti-tools/files/1759631/23_5_east_5c.ts.xz.zip

andimik, 2018-05-28:
5 should be 2B, when you count 0-3 for A, then 4 is 1B and 5 is 2B.

mpbraendli, 2018-05-28:
@andimik I don't think we count 0-3 for EEP-A and 4-7 for EEP-B, this was done like that before. But you are maybe pointing to the problem. I'll check if `FIBProcessor::HandleFIG0Extension1` sets the correct `protOption` in all cases.

mpbraendli, 2018-05-28:
c7b3157 should fix EEP-B handling. Please confirm it works with the ensemble that failed before!

Thanks for reporting the issue, if it persists, please re-open the issue.

closed 2018-05-28 by mpbraendli

r0k, 2018-05-28:
Commit works. Both channels are now listed with bitrate 64. No runtime_error thrown anymore.

There is no sound when selecting one of those channels which somehow relates to #262 but this is another story. Thanks for the fast reply - I will delete the IQ file generated :)

andimik, 2018-05-28:
Good to hear I was having the right idea...