AlbrechtL/welle.io #262: One and the same radio program in 2 ensembles

dabfriend, 2018-05-28:
Schwarzwaldradio (Baden-Württemberg in Germany) is on air at channel 5C and 11B.

The bitrates vary minimal.
Both ones are stored correctly after channel search in opposite to my table radio, where only channel 5 is stored.
If you switch this program from channel 5C to 11B (or the other way round) channel and frequency changes, but not the bitrate and there is no sound from the speakers.
Nothing changes if you switch back.

The only solution is to klick any other station at first and then change Schwarzwaldradio from one channel to the other.

AlbrechtL, 2018-06-24:
How do you recommend to solve this issue?

mpbraendli, 2018-08-28:
@AlbrechtL how does the GUI store the list of programmes? Is it only programme names? Or is it a pair of programme name and EId?

AlbrechtL, 2018-08-30:
In the past it was the program name and the channel name. In the next branch I changed it into the service ID. If both programs are having the same service ID than only one is stored now.

AlbrechtL, 2018-08-30:
@ghost Can you try the latest next [build please](https://bintray.com/albrechtl/welle.io/welle.io_next_nightly#files)