AlbrechtL/welle.io #263: next branch doesn't compile on Debian Sid

russel, 2018-05-28:
The next branch 9ecc4e10288425e8dda55a5e435849e33ec30b38 fails to compile on Debian Sid with gcc version 7.3.0 (Debian 7.3.0-19) 
```
|> ninja
[7/92] Building CXX object CMakeFiles/welle-cli.dir/src/welle-cli/tests.cpp.o
FAILED: CMakeFiles/welle-cli.dir/src/welle-cli/tests.cpp.o 
/usr/bin/c++  -DGITHASH=\"9ecc4e1\" -DHAVE_ALSA -DHAVE_SOAPYSDR -Iwelle-cli_autogen/include -I/home/users/russel/Repositories/Git/Forks/WelleIo/src -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/output -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/various -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/input -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/gui -fvisibility-inlines-hidden -O3 -DNDEBUG   -Wall -g -Wall -Wextra -fvisibility=hidden -std=gnu++14 -MD -MT CMakeFiles/welle-cli.dir/src/welle-cli/tests.cpp.o -MF CMakeFiles/welle-cli.dir/src/welle-cli/tests.cpp.o.d -o CMakeFiles/welle-cli.dir/src/welle-cli/tests.cpp.o -c /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.cpp
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.cpp: In member function ‘virtual int32_t ChannelSimulator::getSpectrumSamples(DSPCOMPLEX*, int32_t)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.cpp:95:66: error: no matching function for call to ‘CVirtualInput::getSpectrumSamples(DSPCOMPLEX*&, int32_t&)’
             { return parentInput->getSpectrumSamples(buffer, size); }
                                                                  ^
In file included from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/CVirtualInput.h:34:0,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/CInputFactory.h:34,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.h:33,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.cpp:33:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-controller.h:164:25: note: candidate: virtual int32_t InputInterface::getSpectrumSamples(std::vector<std::complex<float> >&, int32_t)
         virtual int32_t getSpectrumSamples(std::vector<DSPCOMPLEX> &sampleBuffer, int32_t size) = 0;
                         ^~~~~~~~~~~~~~~~~~
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-controller.h:164:25: note:   no known conversion for argument 1 from ‘DSPCOMPLEX* {aka std::complex<float>*}’ to ‘std::vector<std::complex<float> >&’
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.cpp: In function ‘void test0_iteration(std::unique_ptr<CVirtualInput>&, double)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.cpp:196:22: error: cannot declare variable ‘s’ to be of abstract type ‘ChannelSimulator’
     ChannelSimulator s(interface, stddev);
                      ^
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.cpp:49:7: note:   because the following virtual functions are pure within ‘ChannelSimulator’:
 class ChannelSimulator : public CVirtualInput
       ^~~~~~~~~~~~~~~~
In file included from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/CVirtualInput.h:34:0,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/CInputFactory.h:34,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.h:33,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/tests.cpp:33:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-controller.h:164:25: note: 	virtual int32_t InputInterface::getSpectrumSamples(std::vector<std::complex<float> >&, int32_t)
         virtual int32_t getSpectrumSamples(std::vector<DSPCOMPLEX> &sampleBuffer, int32_t size) = 0;
                         ^~~~~~~~~~~~~~~~~~
[15/92] Building CXX object CMakeFiles/welle-cli.dir/src/welle-cli/webradiointerface.cpp.o
FAILED: CMakeFiles/welle-cli.dir/src/welle-cli/webradiointerface.cpp.o 
/usr/bin/c++  -DGITHASH=\"9ecc4e1\" -DHAVE_ALSA -DHAVE_SOAPYSDR -Iwelle-cli_autogen/include -I/home/users/russel/Repositories/Git/Forks/WelleIo/src -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/output -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/various -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/input -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/gui -fvisibility-inlines-hidden -O3 -DNDEBUG   -Wall -g -Wall -Wextra -fvisibility=hidden -std=gnu++14 -MD -MT CMakeFiles/welle-cli.dir/src/welle-cli/webradiointerface.cpp.o -MF CMakeFiles/welle-cli.dir/src/welle-cli/webradiointerface.cpp.o.d -o CMakeFiles/welle-cli.dir/src/welle-cli/webradiointerface.cpp.o -c /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/webradiointerface.cpp
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/webradiointerface.cpp: In member function ‘bool WebRadioInterface::send_spectrum(Socket&)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/webradiointerface.cpp:817:76: error: no matching function for call to ‘CVirtualInput::getSpectrumSamples(DSPCOMPLEX*&, int16_t&)’
     size_t samples = input.getSpectrumSamples(spectrumBuffer, dabparams.T_u);
                                                                            ^
In file included from /home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-receiver.h:38:0,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/webradiointerface.h:36,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/webradiointerface.cpp:35:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-controller.h:164:25: note: candidate: virtual int32_t InputInterface::getSpectrumSamples(std::vector<std::complex<float> >&, int32_t)
         virtual int32_t getSpectrumSamples(std::vector<DSPCOMPLEX> &sampleBuffer, int32_t size) = 0;
                         ^~~~~~~~~~~~~~~~~~
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-controller.h:164:25: note:   no known conversion for argument 1 from ‘DSPCOMPLEX* {aka std::complex<float>*}’ to ‘std::vector<std::complex<float> >&’
[16/92] Building CXX object CMakeFiles/welle-cli.dir/src/welle-cli.cpp.o
ninja: build stopped: subcommand failed.
```

mpbraendli, 2018-05-28:
@AlbrechtL I'll take care of it

AlbrechtL, 2018-05-28:
I experimenting with references instead of pointers. Do you think this is a good idea `getSpectrumSamples(std::vector<DSPCOMPLEX> &sampleBuffer, int32_t size)` ?

mpbraendli, 2018-05-28:
Yes it's a good thing, but it is better to return the vector. I'll propose something in the commit I'm preparing.

mpbraendli, 2018-05-28:
I'm also not sure if the functions that return rvalue-references do the expected thing, because maybe this makes return-value optimisation impossible. I'll replace those, i.e. :

```
-std::vector<float> &&CRadioController::getImpulseResponse()
+std::vector<float> CRadioController::getImpulseResponse()
 {
     std::lock_guard<std::mutex> lock(impulseResponseBufferMutex);
-    return std::move(impulseResponseBuffer);
+    auto buf = std::move(impulseResponseBuffer);
+    return buf;
 }
```

mpbraendli, 2018-05-28:
Commit 66f964b fixes compilation and replaces VLAs by vectors.

Commit c2182d6 is the rework of the four getter functions for the graphical display.

I'll now merge branch `mpb` into `next`, I think we can work together in the same branch again. I've fixed quite a few new warnings that came up with gcc 8, it would be nice to have them in `next` too.

mpbraendli, 2018-05-28:
Merged and deleted `mpb`. @russel please check if compilation is fine, and close the issue if it's ok!

russel, 2018-05-29:
@mpbraendli  @AlbrechtL  I am packed up and driving today (why a 140 mile journey takes 6 hours is a long story, totally uninteresting to all except myself and my wife :-) so I can only get to this tonight. But I'll get it done.

russel, 2018-05-30:
OK so half a day late. Code duly compiled successfully on Debian Sid – and welle-io also runs. The changeset compiled was eeabf90e7a2d34151dc9ed2e24dea94c12eef04a