AlbrechtL/welle.io #265: Add support for MPRIS for Linux

minecraft2048, 2018-05-30:
On Android, welle.io have those prev/next/play buttons to go to previous station/next station/stop and play, and on the notification display it shows the radio station name and whatever other radio information. On desktop Linux, [MPRIS](https://specifications.freedesktop.org/mpris-spec/latest/) does the same thing, like this 
[image]

AlbrechtL, 2018-06-24:
Thanks for the idea! I added it to the [open task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

closed 2018-06-24 by AlbrechtL

tenzap, 2020-04-11:
see PR #543 
Until it is merged, you can test it through my [fork](https://github.com/tenzap/welle.io/tree/next)