AlbrechtL/welle.io #272: [Windows, next-nightly] builds

beta-tester, 2018-07-01:
you are publishing (kind of) nightly builds of the master branch [here](https://bintray.com/albrechtl/welle.io/welle.io_nightly)

would it be possible to do it as well for the next-nightly [here](https://bintray.com/albrechtl/welle.io/welle.io_next_nightly)?
i am only able to compile welle from source code for linux sucessfully. on the other PC i use Windows, andthere i don't have the abillity to compule it by my self. so it would be usefull to have a **next**-nightly build

AlbrechtL, 2018-07-05:
We had some issues with BinTray because we used to much space there. I had to delete all next builds to make new uploads possible. Please check it out again (https://bintray.com/albrechtl/welle.io/welle.io_next_nightly#files).

closed 2018-07-05 by AlbrechtL

andimik, 2018-07-05:
Is this the next? Because the filenames don't mention it.

beta-tester, 2018-07-06:
yes, thank you there is a file now.

@andimik , yes it is the build of the branch **next**.
the name next can be seen in the link path welle.io_**_next_**_nightly.

andimik, 2018-07-06:
Ok thanks, I didn't double check with the commit version