AlbrechtL/welle.io #274: std::__replacement_assert in src/backend/eep-protection.cpp:131

meeuw, 2018-07-01:
I've compiled welle.io using the Fedora CFLAGS which contains some harderning:
```
export CFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
export CXXFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
export FFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
export FCFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
export LDFLAGS='-Wl,-z,relro  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld'
```

When I select a radio station in welle.io and just before any audio is outputted I get:
```
/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::
size_type) [with _Tp = short int; _Alloc = std::allocator<short int>; std::vector<_Tp, _Alloc>::reference = short int&; std::vector<_Tp, _All
oc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.

Thread 15 "QThread" received signal SIGABRT, Aborted.
```
GDB Backtrace:
```
(gdb) bt
#0  0x00007ffff47c4f2b in raise () from /lib64/libc.so.6
#1  0x00007ffff47af561 in abort () from /lib64/libc.so.6
#2  0x00005555555735da in std::__replacement_assert (__condition=0x5555555b31a0 "__builtin_expect(__n < this->size(), true)", __function=<synthetic pointer>,
    __line=932, __file=0x5555555b31d0 "/usr/include/c++/8/bits/stl_vector.h") at /usr/include/c++/8/x86_64-redhat-linux/bits/c++config.h:2389
#3  std::vector<short, std::allocator<short> >::operator[] (__n=<optimized out>, this=0x7fff9402b0a0) at /usr/include/c++/8/bits/stl_vector.h:932  
#4  eep_protection::deconvolve (this=0x7fff9402ac30, v=0x7fff9c854030, size=<optimized out>, outBuffer=0x7fff94006220 "")
    at /home/meeuw/git/welle.io/src/backend/eep-protection.cpp:131
#5  0x0000555555579ae1 in dabAudio::run (this=0x7fff94004340) at /home/meeuw/git/welle.io/src/backend/audio/dab-audio.cpp:163
#6  0x00007ffff554b11d in QThreadPrivate::start(void*) () from /lib64/libQt5Core.so.5
#7  0x00007ffff6ace594 in start_thread () from /lib64/libpthread.so.0
#8  0x00007ffff488802f in clone () from /lib64/libc.so.6
```

I'm using welle.io version 1.0 (66c48b7) using librtldab.

mpbraendli, 2018-07-02:
There are a few strange things I saw when compiling with `-fsanitize=address` that come from QT, but this one looks different.

Can you try the current `next` branch instead of v1.0 and report your results?

meeuw, 2018-07-02:
Sure, I've compiled welle.io using the `next` branch and then this issue is fixed!

Will the `next` branch be released soon or would it be faster to backport this fix to v1.0?

mpbraendli, 2018-07-02:
Good to hear!

My preference would be more towards "release early, release often", but we have a couple of regressions to fix first.

@AlbrechtL is working on a GUI redesign that needs to be completed before a release, and the Android build is also broken (probably). I intend to conclude the multipath reception improvements in the backend.

This page gives an idea: https://github.com/AlbrechtL/welle.io/projects/1
We would welcome some help, have a look at the tickets in "To do" if you have time and let us know!

So I guess the answer to your question is "release soon" with an undefined meaning of the word "soon" :-)

AlbrechtL, 2018-07-06:
@meeuw You are welcome to join the welle.io development team if you like to speed up the next release.

closed 2018-07-06 by AlbrechtL