AlbrechtL/welle.io #277: Mono stations are shown as stereo

andimik, 2018-07-17:
In Slovenija, some stations use 32 kHz 24 kBit mono, but it is shown as stereo.

[image]

Tested with b68d488 . 

[slo_10d_2018-07-08_1146.eti.zip](https://github.com/AlbrechtL/welle.io/files/2203424/slo_10d_2018-07-08_1146.eti.zip)

`cat slo_10d_2018-07-08_1146.eti | odr-dabmod eti2raw.ini | ./20180717_b68d488_Linux_welle-io-x86_64.AppImage --raw-file /dev/stdin  -d rawfile`

and eti2raw.ini is

```
[input]
transport=file
source=/dev/stdin

[modulator]
gainmode =1
rate=2048000

[output]
output=file

[fileoutput]
format=u8
filename=/dev/stdout

```

Dablin shows mono (and I know it as well)

[image]

AlbrechtL, 2018-07-19:
Can it be that this station uses parametric stereo?

basicmaster, 2018-07-19:
No, as in such a case DABlin indicates "HE-AAC v2" and "Stereo".

andimik, 2018-07-20:
No, this is mono.

AlbrechtL, 2018-07-20:
OK then, it seems to be a bug.

mpbraendli, 2018-07-23:
Wow, your ETI file has a lot of frame count discontinuities! I think I'll generate an ETI with all combinations to test.

andimik, 2018-07-23:
I see, I will check with another one

mpbraendli, 2018-07-23:
Fixed in 0cac18f

And I discovered that `welle-cli` cannot handle mono streams at all :-/ Another thing to debug...

@andimik please open the issue again if you still have a problem with the signalling of mono/stereo with your programme!

closed 2018-07-23 by mpbraendli

andimik, 2018-07-23:
Merci :-)

mpbraendli, 2018-07-23:
welle-cli mono decoding is now also fixed.

AlbrechtL, 2018-07-23:
Makes it sense to print the AAC mode in welle-io in expert mode as well?

mpbraendli, 2018-07-23:
I believe it is. This idea is tracked in https://github.com/AlbrechtL/welle.io/projects/1#card-9663493 already.