AlbrechtL/welle.io #280: No DLS at certain DAB channels from UK

andimik, 2018-08-07:
While investigating a problem with qt-dab (see https://github.com/JvanKatwijk/qt-dab/issues/107 ), I found out by chance, that these DAB channels from GB

- "Panjab Radio    " ( CED8 )
- "Premier Praise  " ( CDD8 )
- "PremierChristian" ( CCD8 )
- "Sunrise Radio UK" ( C9D8 )
- "talkRADIO       " ( C0D8 )
- "talkSPORT 2     " ( C1D8 )

 don't show DLS in welle.io, but in qt-dab and XPadXpert they do so ...

Screenshots taken with b68d488

Here is an ETI file which you can use

[sdl_short.eti.zip](https://github.com/AlbrechtL/welle.io/files/2267924/sdl_short.eti.zip)

`cat sdl_short.eti | odr-dabmod ini-file.ini | 20180717_b68d488_Linux_welle-io-x86_64.AppImage` 

whereas the `ini-file.ini` is 

```
[input]
transport=file
source=/dev/stdin

[modulator]
gainmode =1
rate=2048000

[output]
output=file

[fileoutput]
format=u8
filename=/dev/stdout
```

[image]

[image]

[image]

AlbrechtL, 2018-08-09:
Which welle.io version did you use?

andimik, 2018-08-09:
See my text which shows the version as well :blush:

> Screenshots taken with b68d488

AlbrechtL, 2018-08-09:
Can you try it with DABlin, please?

andimik, 2018-08-09:
Could you do it? The eti file is attached? I am on vacation

basicmaster, 2018-08-10:
With DABlin, DL is shown for all of the mentioned services.

sheimers, 2018-08-12:
Please also check if the Xpad type in the ETI file is correct. There can be a misconfiguration at the multiplexer or a bug in the multiplexer software, that happened before and some receivers ignore it and work fine while others refuse. So the problem can be either in welle.io or at the transmitter.

sheimers, 2018-08-12:
I think X-Pad Application type should usually be 12, but one of the commercial multiplexers sets it to 2 in stead of 12 if "MOT & DLS" is selected in the configuration GUI. See page 20 of https://www.etsi.org/deliver/etsi_en/301200_301299/301234/01.01.01_60/en_301234v010101p.pdf

basicmaster, 2018-08-12:
MOT and DLS are transmitted separate from each other in PAD, using different Application Types. The Application Types 2/3 used here for DLS are correct and absolutely fine. Any MOT can be ignored here.

AlbrechtL, 2018-08-15:
@andimik Can you provide a RAW file, please?

andimik, 2018-08-15:
No, please use the eti and create it yourself

AlbrechtL, 2018-08-16:
Please help us to analyze the issue by providing a RAW file. This is the most easiest way to reproduce the issue by us. Also it doesn't matter if those RAW files are existing in the filed, it is just for debugging.

andimik, 2018-08-16:
No, I won't. Use ODR-DabMod for that reason. I don't receive the DAB signal, only the feed (which is eti).

All other users do it this way.

mpbraendli, 2018-08-28:
@andimik please understand not everyone has ODR-DabMod installed just for such situations. Both @AlbrechtL have lots of things to do, and if we want to make progress, we need your help.

It's good to report issues, but it's better to report issues we can reproduce easily, and if it costs a bit of internet bandwidth, that's not a big deal.

Since you already have everything in place to make an .iq file, please do so, it helps us.

andimik, 2018-08-30:
@AlbrechtL I really recommend you to install ODR-DabMod. The other tools from Opendigitalradio are not necessary at the moment.

But you have to understand, the issue is well (!) documented in my first posting.

Here you find the sdl_short.eti as raw file ...
https://we.tl/t-OSlC2MjQVh
**(188 Megabyte zipped RAW-file compared to 9,7 MB for the same ETI file)**

Tested with 20180830_ec3eeff_Linux_welle-io-x86_64.AppImage

AlbrechtL, 2018-08-31:
Thanks for the file.

mpbraendli, 2018-08-31:
> But you have to understand, the issue is well (!) documented in my first posting.

Sorry if I insist again, but the point is not that your issue wasn't "well documented". "Easily reproducible" is what we're asking for.

Thanks for the IQ file.

AlbrechtL, 2018-08-31:
Short update on the ongoing investigation. It seems that this bug has something to do with the 24 kHz sampling frequency. All channels with 48 kHz seems to be OK. Maybe some length calculations are wrong.

AlbrechtL, 2018-09-01:
@basicmaster  Similar code can be found here: https://github.com/AlbrechtL/welle.io/blob/5fb8939207934d5e2795a2884422a800ae5520d0/src/backend/audio/mp2processor.cpp#L643

I'm now thinking if we take the opportunity to switch your dablin DAB decoder which is MPG123 based.

basicmaster, 2018-09-07:
This sounds like a good idea. Let me know if this would require any adjustments in `dabdecoder.cpp/h`. It makes sense to then generalise the code also in DABlin (as recently done with the PAD encoder), to ease later updates.

At least I would in general recommend to switch to libmpg123, to have a clear abstraction between the decoding and welle.io, and also benefit from this lib's ongoing development (though it is of course already mature).

AlbrechtL, 2018-09-12:
Fixed in https://github.com/AlbrechtL/welle.io/commit/a8391cbe80b5f41fd22c8aafdfc51e0ef370ee4c.
Now, welle-cli shows all dynamic labels
[image]

As well as welle.io
[image]

@basicmaster In dab_decoder.cpp I had to switch from float32 samples to int16 samples.

closed 2018-09-12 by AlbrechtL

basicmaster, 2018-09-12:
@AlbrechtL:
I just modified my code to have a constructor option to select between int16 and float32, and added the missing `<vector>` include to the header. So you should now be able to use the unmodified versions of `dab_decoder.*`.

Besides the initial "MP2Decoder: using decoder" message, the only console output is done when a CRC error is detected and hence the current frame is discarded: `(CRC)`

Would it make sense to instead forward this information to the user, similar to the X-PAD length issue indication?

AlbrechtL, 2018-09-13:
From my point of view to output messages directly from one module is not a big deal. But I would appreciate if you switch from `fprintf(stderr...)` to `std::clog` like [here](https://github.com/AlbrechtL/welle.io/blob/next/src/backend/mp2processor.cpp#L75).
The reason is that welle.io GUI redirects the clog output direct into the GUI. Redirecting stderr and stdout under Windows is really hard.

For the X-PAD length it is also OK to create a message.

@mpbraendli What do you think?

AlbrechtL, 2018-09-13:
Lets use the new issue https://github.com/AlbrechtL/welle.io/issues/302 to discuss the code integration more further.

mpbraendli, 2018-09-13:
Good idea to create a API function for the CRC error, similar to how the X-PAD length error is signalled.