AlbrechtL/welle.io #282: [next] How to present TII information inside the GUI?

AlbrechtL, 2018-08-09:
This issue is to discuss how to present the TII information inside the GUI.

At the moment the TII is only printed into the standard output.
[image]

mpbraendli, 2018-08-28:
Currently, the information can change quite often, and false TII comb pattern pairs can show up in the list if the receiver is running for a longer duration. This has to be fixed because the measurement algorithm is missing a good quality criterion.

It would be nice to display this TII information alongside the CIR, maybe even on the same plot. I don't know if it's easy to find out which CIR peak corresponds to which TII, which would be the nicest way to display.

AlbrechtL, 2018-08-30:
Do you mean to display the text inside the plot? Can you provide a simple mockup?

mpbraendli, 2018-09-04:
Here's a mockup with a base picture from welle-cli, but the idea is the same

[image]

The green bars correspond to where we think the TII is relative to the CIR graph. Wide bars mean there is less confidence. The number next to it is mainId:subId.

But before we can do this, we need

* a better working confidence indication
* calculate absolute delays relative to start of NULL symbol (or start of PRS) and align this with the CIR calculation

What do you think about such a representation?

AlbrechtL, 2018-09-05:
> What do you think about such a representation?

It looks great! But I don't know how difficult the graphic implementation is.

AlbrechtL, 2018-11-01:
To add the TII information to the CIR is not really complicated (https://github.com/AlbrechtL/welle.io/issues/309) as long we know at which sample which information needs to be shown. Is this information is available?

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL