AlbrechtL/welle.io #283: wrong characters for DLS in polish language, old EBU table?

sheimers, 2018-08-12:
For a test I tried to transmit words with characters from eastern European languages in the DLS. The string was "äöü ÄÖÜ ß €$ ěřčČůÚšŠžŽ ğçş Antonín Dvořák Český rozhlas Živé vysílání Lech Wałęsa Doviđenja!"

As it turned out most languages seem to work fine except polish where "Wałęsa" was not correct. See the attached screenshot of welle.io on my Android phone.

[image]

My guess is that it is using an old version of the EBU latin character table. The current table can be found in the source code of ODR-PadEnc or in the official EBU standard. See:

https://github.com/Opendigitalradio/ODR-PadEnc/blob/master/src/charset.cpp

https://www.etsi.org/deliver/etsi_ts/101700_101799/101756/01.08.01_60/ts_101756v010801p.pdf

sheimers, 2018-08-14:
I found the EBU character table is in src/backend/charsets.cpp and is indeed the old version 1.6.1. If nobody else wants to do it I can offer to update the table myself within two or three days from now.

mpbraendli, 2018-08-16:
Thanks for the report and bugfix! I'll close this issue, as you opened a pull request fixing this.

closed 2018-08-16 by mpbraendli