AlbrechtL/welle.io #286: Request to add recording functionality

diddy12345, 2018-08-17:
Could I request to add a recording feature to Welle.io ?

This could be a simple record button or maybe even a timed recording feature where the user can set a recording time and the station to record from.

If there are legal issue, I suspect this can be worked around. Other software like SDR Touch allows recording

AlbrechtL, 2018-08-17:
What do you mean with recording? IQ RAW recordings or audio recordings?

diddy12345, 2018-08-20:
audio recordings. I couldn't see much use recording the iq raw unless a user would want to record the whole mux but i dont think there is enough bandwidth to do that (6mhz)

AlbrechtL, 2018-08-20:
Please see #239. You may can use welle-cli.

AlbrechtL, 2018-08-20:
The general discussion about recording can he found here: #36

closed 2018-08-20 by AlbrechtL