AlbrechtL/welle.io #289: "auto" language setting is annoying

snowflakeas, 2018-08-26:
Hi - first time user. My Win10 install is English (Norwegian keyboard). 
Welle autoinstalls as Norwegian (whereas having an English Version of Windows, I obviously want English as my language).

I cannot find the setting in the windows Registry either.

Possible to add query during install, or find a better way to get language setting?

AlbrechtL, 2018-08-27:
How your other programs behaves?

Regarding LimeSDR: Your only option is to use Linux and to compile welle.io from source.

mpbraendli, 2018-08-28:
Other topic: if you manage to get SoapySDR running on windows, please tell us how it's done in the wiki! But you will need to compile everything yourself with CMake. https://github.com/pothosware/SoapySDR/wiki/BuildGuide#windows

snowflakeas, 2018-08-28:
I just ran the executable install from LimeSDR mini (myriad)

closed 2018-08-28 by snowflakeas