AlbrechtL/welle.io #290: [next] welle-cli -Dw does not work with data streams (packetdata)

andimik, 2018-08-27:
If I load a file containing one or more data streams, the option `-Dw` does not work:

```
$ ./welle-cli -Dw 7979 -f ~/stdin.raw
Hello this is welle-cli V1.0-426-g9c543d4
RAWFilePeriod =8000
OFDM-processor:start
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910001 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
Tune to 0x0xe4910002 failed
```

[welle_cli.pdf](https://github.com/AlbrechtL/welle.io/files/2324035/welle_cli.pdf)

I've used Slovenian DAB in this case, see PDF attached. They have one EPG and one Journaline service.

A stream without Data stream works, on the other hand, see the next PDF.

[welle_cli_hrv.pdf](https://github.com/AlbrechtL/welle.io/files/2324047/welle_cli_hrv.pdf)
(ignore the errors, the eti file which was processed by odr-dabmod was too short)

mpbraendli, 2018-08-28:
I can also receive such an ensemble here. It should be easy to fix

mpbraendli, 2018-09-04:
Fixed in latest `next`

closed 2018-09-04 by mpbraendli

andimik, 2018-09-04:
I'll try in the next days. Thanks anyway.