AlbrechtL/welle.io #295: [next] welle-cli: building fails due to mm_malloc.h

basicmaster, 2018-09-02:
System is Ubuntu 16.04 AMD64; I tried to build welle.io-cli using CMake with `-DRTLSDR=1 -DBUILD_WELLE_IO=OFF`.

However when `/src/backend/ofdm/phasereference.cpp` is to be build, the compilation aborts:

```
In file included from /home/stefan/DL-Source/welle.io/src/backend/mm_malloc.h:45:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include/xmmintrin.h:34,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include/x86intrin.h:31,
                 from /usr/include/x86_64-linux-gnu/c++/5/bits/opt_random.h:33,
                 from /usr/include/c++/5/random:50,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.cpp:23:
/usr/include/malloc.h:71:29: error: ‘ptrdiff_t’ was not declared in this scope
 extern void *(*__morecore) (ptrdiff_t __size);
                             ^
/usr/include/malloc.h:71:29: note: suggested alternatives:
In file included from /usr/include/c++/5/cstdint:38:0,
                 from /home/stefan/DL-Source/welle.io/src/backend/charsets.h:27,
                 from /home/stefan/DL-Source/welle.io/src/backend/dab-constants.h:37,
                 from /home/stefan/DL-Source/welle.io/src/various/fft.h:34,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.h:26,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.cpp:21:
/usr/include/x86_64-linux-gnu/c++/5/bits/c++config.h:197:28: note:   ‘std::ptrdiff_t’
   typedef __PTRDIFF_TYPE__ ptrdiff_t;
                            ^
/usr/include/x86_64-linux-gnu/c++/5/bits/c++config.h:197:28: note:   ‘std::ptrdiff_t’
In file included from /usr/include/fftw3.h:101:0,
                 from /home/stefan/DL-Source/welle.io/src/various/fft.h:45,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.h:26,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.cpp:21:
/usr/lib/gcc/x86_64-linux-gnu/5/include/stddef.h:149:26: note:   ‘fft::ptrdiff_t’
 typedef __PTRDIFF_TYPE__ ptrdiff_t;
                          ^
In file included from /home/stefan/DL-Source/welle.io/src/backend/mm_malloc.h:45:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include/xmmintrin.h:34,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include/x86intrin.h:31,
                 from /usr/include/x86_64-linux-gnu/c++/5/bits/opt_random.h:33,
                 from /usr/include/c++/5/random:50,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.cpp:23:
/usr/include/malloc.h:74:34: error: ‘ptrdiff_t’ was not declared in this scope
 extern void *__default_morecore (ptrdiff_t __size)
                                  ^
/usr/include/malloc.h:74:34: note: suggested alternatives:
In file included from /usr/include/c++/5/cstdint:38:0,
                 from /home/stefan/DL-Source/welle.io/src/backend/charsets.h:27,
                 from /home/stefan/DL-Source/welle.io/src/backend/dab-constants.h:37,
                 from /home/stefan/DL-Source/welle.io/src/various/fft.h:34,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.h:26,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.cpp:21:
/usr/include/x86_64-linux-gnu/c++/5/bits/c++config.h:197:28: note:   ‘std::ptrdiff_t’
   typedef __PTRDIFF_TYPE__ ptrdiff_t;
                            ^
/usr/include/x86_64-linux-gnu/c++/5/bits/c++config.h:197:28: note:   ‘std::ptrdiff_t’
In file included from /usr/include/fftw3.h:101:0,
                 from /home/stefan/DL-Source/welle.io/src/various/fft.h:45,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.h:26,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.cpp:21:
/usr/lib/gcc/x86_64-linux-gnu/5/include/stddef.h:149:26: note:   ‘fft::ptrdiff_t’
 typedef __PTRDIFF_TYPE__ ptrdiff_t;
                          ^
In file included from /usr/include/features.h:367:0,
                 from /usr/include/x86_64-linux-gnu/c++/5/bits/os_defines.h:39,
                 from /usr/include/x86_64-linux-gnu/c++/5/bits/c++config.h:482,
                 from /usr/include/c++/5/cstdint:38,
                 from /home/stefan/DL-Source/welle.io/src/backend/charsets.h:27,
                 from /home/stefan/DL-Source/welle.io/src/backend/dab-constants.h:37,
                 from /home/stefan/DL-Source/welle.io/src/various/fft.h:34,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.h:26,
                 from /home/stefan/DL-Source/welle.io/src/backend/ofdm/phasereference.cpp:21:
/usr/include/malloc.h:75:1: error: expected ‘,’ or ‘;’ before ‘throw’
 __THROW __attribute_malloc__;
 ^
```

In the end the reason is that the project contains a file called `mm_malloc.h`. This filename however is also used by the system, and (apparently due to the order of include paths or so) the project's file is included instead of the system level file (due to `#include <mm_malloc.h>` in `/usr/lib/gcc/x86_64-linux-gnu/5/include/xmmintrin.h`).

When I remove the big `#ifdef` block from the project's `mm_malloc.h`, the complation works fine. I wonder why a custom `malloc` is needed here...

andimik, 2018-09-03:
Under 18.04 I had no problem.

mpbraendli, 2018-09-04:
Oh I didn't see that the spiral viterbi code (which was removed) brought in this header file. That's doesn't look proper to me...

I'll change this and see if it breaks the windows build. I don't know how aligned memory allocation should be done there. But since the `MALLOC` macro isn't used anywhere it might not change a thing.

mpbraendli, 2018-09-04:
@basicmaster does latest `next` now compile?

basicmaster, 2018-09-04:
Yes, this fixes it for me. Thanks!

closed 2018-09-04 by mpbraendli