AlbrechtL/welle.io #296: [welle-cli] Crash on illegal UTF-8

gvanem, 2018-09-05:
Running `welle-cli` on Windows crashes deep inside `WebRadioInterface::serve()` for no obvious reason. Tested with Google Chrome and `welle-cli.exe -C1 -w 1212 -c 12B`.

After adding a `try / catch` block around `fut.get()` inside `WebRadioInterface::serve()`, I can see the error is:
```
Exception: [json.exception.type_error.316] invalid UTF-8 byte at index 6: 0x72
```
Lots of these. But what are they and why? Could it be one of these:
  * `s.serviceLabel.utf8_label()`
  * `s.serviceLabel.utf8_shortlabel()`

since if I run `welle-cli.exe -c 12B -D`, there are these services with seemingly illegal UTF-8 encoding:
```
Service list
**** Enter '.' to quit.
New Service: 0xf304 'NRK Vµr Vest    '
...
New Service: 0xf205 'NRK S┴PMI       '
```

mpbraendli, 2018-09-05:
Mmmmh weird... It must be either Labels or DLS. What character encoding is this ensemble using?

Does it crash if you run the receiver without accessing the webpage? If yes, please run the receiver, and execute `wget http://localhost:1212/fic`, let it run for a minute, Ctrl-C it, and send me the file with the FIC.

gvanem, 2018-09-05:
> What character encoding is this ensemble using?

No idea really (ISO-8859-2 from the looks of the debug). How can figure out that?

> Does it crash if you run the receiver without accessing the webpage?

No.

andimik, 2018-09-05:
Can you try it with 12D, Riks? 

https://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=NOR&inputEn=F201

They don't have Norwegian characters in the label.

mpbraendli, 2018-09-05:
> No idea really (ISO-8859-2 from the looks of the debug). How can figure out that?

It's easiest, both for you and for me, to make the FIC dump I mentioned earlier.

I wrote "if yes" in my previous message, but I actually meant "if no" (because it wouldn't make sense otherwise :-D)

Can you send me a one minute FIC dump?

gvanem, 2018-09-05:
Same thing, but at a different offset:
```
Exception: [json.exception.type_error.316] invalid UTF-8 byte at index 10: 0x3A
```
But with `welle-cli.exe -CN -w 1212 -c 12D`, I do get sound and a Audio bar:
[image]

The DLS text looks fine. But the json exception is maybe related to some problem in `charsets.cpp`?

mpbraendli, 2018-09-05:
Yes thanks. That file somehow makes etisnoop segfault, but I can probably analyse something already.

mpbraendli, 2018-09-05:
I am confident it's using Charset 0, i.e. EBU-Latin.

mpbraendli, 2018-09-05:
I ran a test with a custom ensemble with labels containing accents, it seems to work with welle-cli: 
[image]

It's also using charset 0.

Maybe there is one character in a label that is not correctly decoded? I can check this using your FIC dump.

gvanem, 2018-09-05:
> It's also using charset 0.

IOW `CharacterSet::EbuLatin`. I've being trying to understand what the `toUtf8StringUsingCharset()` function does. The default parameter `num_bytes = 0` and the comment says:
```
 * If num_bytes is nonzero, the buffer must be zero
 * terminated.
```
Then why is this:
```c
                if (num_bytes == 0) {
                    while (*buf) {
                        s += ebuLatinToUcs2[*buf];
                        buf++;
                    }
                }
```
assuming `buf` is 0-terminated? That's the opposite of what the comment says. Is the comment wrong?

gvanem, 2018-09-13:
> the browser window does not show anything.

`welle-cli` needs `index.html` and `index.js` in it's directory. Are they? I do not use the `make install`, so I can't tell if it does the copy.

mpbraendli, 2018-09-13:
Correct. There ought to be a cross platform way to

1. copy `index.html` and `index.js` to the right place on `make install`;
1. tell the `welle-cli` binary where this right place is.

I don't know how to do that.

PS:
you should have opened a new issue, as this is a completely different topic.

andimik, 2018-09-13:
Sure. But I thought I made a mistake ...

AlbrechtL, 2018-09-13:
Is the origin issue still exists? If yes, are they any RAW files to test?

BTW: In the latest Windows installer, welle-cli is included incl. the HTML and JS files. If it is not working as expected please open new issues.

AlbrechtL, 2018-09-19:
Can you reproduce the issue now?

gvanem, 2018-09-24:
> Can you reproduce the issue now?

The problem is still present. Welle-cli should be able to cope with illegal UTF-8 sequences.

mpbraendli, 2018-09-24:
Just to clarify my understanding of the issue, I don't see where these illegal UTF-8 sequences come from. Your ensemble contains EBU Latin encoded labels, and we convert these to UCS-2 using the function you also inspected.

I checked that the function generating UCS-2 generates valid output for all 255 possible input values (EBU Latin, skipping '\0').

It's not clear to me how to proceed. I would suggest you add `try` `catch` blocks in the `send_mux_json` function around those places where the labels are being inserted into the json. Then analyse the byte sequence when something fails.

Or maybe you'll find out that it's not the labels, which would help us further.

andimik, 2018-09-24:
Could you share an eti file?

mpbraendli, 2018-09-24:
Since this is an over-the-air signal, and IQ file would be better, as the regenerated ETI will probably contain errors (like the FIC dump).

AlbrechtL, 2018-12-27:
Any news here? Can we close it?

AlbrechtL, 2019-03-03:
I'm closing it because of inactivity.

closed 2019-03-03 by AlbrechtL

andimik, 2020-11-28:
As I'm now able to receive all NRK DAB+ ensembles via satellite, I've tested it (under Ubuntu). It does not crash, but I have the fic-file for you.

[image]

[nrk2.zip](https://github.com/AlbrechtL/welle.io/files/5611102/nrk2.zip)
(contains `nrk2.fic` file)

andimik, 2020-11-28:
And here there is an eti file (which I was using to produce the u8 stream with **odr-dabmod** and to feed into **welle-cli**)

https://www.file-upload.net/download-14380952/nrk2.eti.zip.html

gvanem, 2020-11-29:
@andimik Chrome says "nrk2.eti.zip can be dangerous. So Chrome has blocked it". 
Besides I've forgot about this issue **and** it's impossible to build `welle-cli.exe` for me now due to C++ errors.
Besides how shall I use these .fic files?

andimik, 2020-11-29:
Then use another browser...

You should only download these two files if you know what to do with them.

andimik, 2020-11-29:
I've tried under Windows 10 now and have no problems with words like Østlands, Vær Sør, SÁPMI or Mjøsbyene.

[image]

[image]

Have you tried to use the setup.exe file from https://bintray.com/albrechtl/welle.io/welle.io_nightly#files ? In the installation folder you will find `c:\Program Files (x86)\welle.io\welle-cli.exe`

AlbrechtL, 2021-03-06:
@andimik Thanks for testing!