AlbrechtL/welle.io #298: [next, Discussion] README.md improvments

AlbrechtL, 2018-09-06:
While trying to build welle-cli under Ubuntu I found that the build process in the README.md is not optimal descried.

* The libmp3lame-dev dependency is missing
* I think also if a user just want to build welle-cli we should add example
`cmake ../ -DBUILD_WELLE_IO=OFF`

My idea is to divide the README.md into a welle-gui part and a welle-cli part. We can also use two separate readme files for that.

mpbraendli, 2018-09-08:
Ok having separate "chapters" in the README. Separate READMEs are ok too, if there's enough discoverability of both if one only reads the main `README.md`.

andimik, 2018-09-13:
Sure. As I cannot help you with your code this is the only chance ...

AlbrechtL, 2019-05-03:
Closed because of no activity.

closed 2019-05-03 by AlbrechtL