AlbrechtL/welle.io #299: [welle-cli] Windows build issues

AlbrechtL, 2018-09-07:
While trying to compile welle-cli under Windows the following compiler errors are occurring.
Build environment is MinGW 5.3 (gcc 5.3.0)

   ````
g++ -c -fno-keep-inline-dllexport -g -std=gnu++1y -Wall -W -Wextra -fexceptions -mthreads -DUNICODE -D_UNICODE -DWIN32 -DGITDESCRIBE=\"6ab13e5\" -DQT_QML_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -DQT_NEEDS_QMAIN -I..\..\..\welle.io_next\src\welle-cli -I. -I..\..\..\welle.io_next\src -I..\..\..\welle.io_next\src\backend -I../audio -I../data -I../ofdm -I..\..\..\welle.io_next\src\various -I..\..\..\welle.io_next\src\input -I..\..\..\welle.io-win-libs\include -IC:\Qt\5.11.1\mingw53_32\include -IC:\Qt\5.11.1\mingw53_32\include\QtGui -IC:\Qt\5.11.1\mingw53_32\include\QtANGLE -IC:\Qt\5.11.1\mingw53_32\include\QtCore -Idebug -IC:\Qt\5.11.1\mingw53_32\mkspecs\win32-g++  -o debug\welle-cli.o ..\..\..\welle.io_next\src\welle-cli\welle-cli.cpp
In file included from C:/Qt/Tools/mingw530_32/i686-w64-mingw32/include/oleidl.h:7:0,
                 from C:/Qt/Tools/mingw530_32/i686-w64-mingw32/include/ole2.h:38,
                 from C:/Qt/Tools/mingw530_32/i686-w64-mingw32/include/wtypes.h:12,
                 from C:/Qt/Tools/mingw530_32/i686-w64-mingw32/include/winscard.h:10,
                 from C:/Qt/Tools/mingw530_32/i686-w64-mingw32/include/windows.h:97,
                 from C:/Qt/Tools/mingw530_32/i686-w64-mingw32/include/winsock2.h:23,
                 from ..\..\..\welle.io_next\src/various/Socket.h:39,
                 from ..\..\..\welle.io_next\src/welle-cli/webradiointerface.h:37,
                 from ..\..\..\welle.io_next\src\welle-cli\welle-cli.cpp:50:
..\..\..\welle.io_next\src/welle-cli/tests.h:39:47: error: expected ',' or '...' before 'struct'
         Tests(std::unique_ptr<CVirtualInput>& interface,
                                               ^
..\..\..\welle.io_next\src/welle-cli/tests.h:49:41: error: expected unqualified-id before 'struct'
         std::unique_ptr<CVirtualInput>& interface;
                                         ^
In file included from ..\..\..\welle.io_next\src\welle-cli\welle-cli.cpp:51:0:
..\..\..\welle.io_next\src/welle-cli/tests.h:49:39: error: expected ';' at end of member declaration
         std::unique_ptr<CVirtualInput>& interface;
                                       ^
..\..\..\welle.io_next\src/welle-cli/tests.h:49:50: error: expected identifier before ';' token
         std::unique_ptr<CVirtualInput>& interface;
                                                  ^
..\..\..\welle.io_next\src\welle-cli\welle-cli.cpp: In function 'int main(int, char**)':
..\..\..\welle.io_next\src\welle-cli\welle-cli.cpp:410:36: error: no matching function for call to 'Tests::Tests(std::unique_ptr<CVirtualInput>&, RadioReceiverOptions&)'
         Tests tests(in, options.rro);
                                    ^
In file included from ..\..\..\welle.io_next\src\welle-cli\welle-cli.cpp:51:0:
..\..\..\welle.io_next\src/welle-cli/tests.h:39:9: note: candidate: Tests::Tests(std::unique_ptr<CVirtualInput>&)
         Tests(std::unique_ptr<CVirtualInput>& interface,
         ^
..\..\..\welle.io_next\src/welle-cli/tests.h:39:9: note:   candidate expects 1 argument, 2 provided
..\..\..\welle.io_next\src/welle-cli/tests.h:37:7: note: candidate: constexpr Tests::Tests(const Tests&)
 class Tests {
       ^
..\..\..\welle.io_next\src/welle-cli/tests.h:37:7: note:   candidate expects 1 argument, 2 provided
..\..\..\welle.io_next\src/welle-cli/tests.h:37:7: note: candidate: constexpr Tests::Tests(Tests&&)
..\..\..\welle.io_next\src/welle-cli/tests.h:37:7: note:   candidate expects 1 argument, 2 provided
Makefile.Debug:697: recipe for target 'debug/welle-cli.o' failed
   ````
Is the gcc to old for that?

gvanem, 2018-09-07:
MinGW (e.g. TDM-MinGW) has this in several headers (including `<basetyps.h>`):
```c
#ifndef __OBJC__
#undef interface
#define interface struct
#endif
```
Try an `#undef interface` after the headers.

AlbrechtL, 2018-09-07:
This was the issue. Thanks!

Fixed in https://github.com/AlbrechtL/welle.io/commit/a53a45b747f4258709b8382b06497f044a6181b0.

closed 2018-09-07 by AlbrechtL