AlbrechtL/welle.io #300: CFaadDecoder: can't write WAV sample to buffer.

frazzmatazz, 2018-09-11:
I've recently compiled Welle.io from source on linux. Runs beautifully, picking up 54 stations, with 3 rectangles and 2 bars (I'll admit my antenna is a bit borderline.) and even pictures I never normally get to see!

But I can't get any audio output. My terminal fills up with:
`CFaadDecoder: can't write WAV sample to buffer. Wrote 0 bytes of 3840 bytes.`
and occasional:
`using null output device, none available`

Pulseaudio's pavucontrol shows QuiskDigitalOutput but no VU meter activity, and nothing on the usual "Built-in Audio Analogue Stereo".

I suspect i missed some important QT audio library when compiling, but I'm stuck despite googling.

frazzmatazz, 2018-09-11:
This reports they are all the latest versions: (Ubuntu 18.04.1 LTS)
`sudo apt install libfaad-dev libfftw3-dev librtlsdr-dev libusb-1.0-0-dev mesa-common-dev libglu1-mesa-dev libpulse-dev
`

The precompiled version does work though.

AlbrechtL, 2018-09-11:
Which Qt version did you use?

frazzmatazz, 2018-09-11:
qmake --version
QMake version 3.1
Using Qt version 5.9.5 in /usr/lib/x86_64-linux-gnu

AlbrechtL, 2018-09-11:
Can you try the latest [next branch](https://github.com/albrechtl/welle.io/tree/next) together with Qt 5.10 please?

alexmyczko, 2018-09-18:
this is what I'm getting:

./welle-io
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
PulseAudioService: pa_context_connect() failed
using null output device, none available
QQmlApplicationEngine failed to load component
qrc:/src/gui/QML/main.qml:42 module "QtQuick.Controls" version 2.0 is not installed

AlbrechtL, 2018-09-19:
@alexmyczko Is your problem solved?

AlbrechtL, 2018-10-25:
Closed because of no progress.

closed 2018-10-25 by AlbrechtL