AlbrechtL/welle.io #302: [Discussion] Integration of code from DABlin

AlbrechtL, 2018-09-13:
This issue is to discuss the ongoing integration of more code from DABlin.
The discussion has started in issue https://github.com/AlbrechtL/welle.io/issues/280

mpbraendli, 2018-09-13:
It would be a shame to lose this information. It's a helpful quality indicator.

AlbrechtL, 2018-09-13:
@basicmaster Is it OK for you to add functions for the errors (CRC, RS etc.)?

basicmaster, 2018-09-13:
> In 66d6982 I switched also to the DAB+ decoder from DABlin (the update to 16-bit samples necessary).

Great! I will adjust the constructor, to provide a param for 16bit int here as well.

> It would be a shame to lose this information. It's a helpful quality indicator.

> @basicmaster Is it OK for you to add functions for the errors (CRC, RS etc.)?

I agree that this information should definitely be available and will modify the involved classes to forward it.

basicmaster, 2018-09-13:
OK; I adjusted the affected classes. You should now able to use them without modifying them. The errors/warnings/infos of the audio decoders are now forwarded to the `SubchannelSinkObserver`.

AlbrechtL, 2018-09-14:
Thanks! I updated the files in https://github.com/AlbrechtL/welle.io/commit/403b331172d98c44c86d243bf22699232f0a2ae6.

@mpbraendli We have to adapt our back-end API to the new behavior.
The frame errors and the AAC errors are gone instead we have an audio error string and an audio warning string
https://github.com/AlbrechtL/welle.io/blob/403b331172d98c44c86d243bf22699232f0a2ae6/src/backend/decoder_adapter.cpp#L124-L132

The RS error are still existing
https://github.com/AlbrechtL/welle.io/blob/403b331172d98c44c86d243bf22699232f0a2ae6/src/backend/decoder_adapter.cpp#L134-L138
@basicmaster In my case `uncorr_errors` was all the time `true`.

basicmaster, 2018-09-14:
BTW you might want to put the printed error/warning hints in quotation marks ;-)

Having good reception, RS errors should only occur during Superframe sync (until the correct one out of five DAB frames has been found), where `uncorr_errors` will always be true, as there is nothing (yet) to be corrected. During playback no RS callbacks should occur with good reception.

During playback with bad reception (e.g. pull in the antenna), at first RS errors should be corrected without `uncorr_errors` being set. So all errors could be corrected in that case. Only if reception gets even worse, some errors will remain uncorrected and `uncorr_errors` will again be true (and also AUs discarded due to wrong CRC).

AlbrechtL, 2018-09-14:
> BTW you might want to put the printed error/warning hints in quotation marks ;-)

Added in https://github.com/AlbrechtL/welle.io/commit/301e1906961b4487ae3107f8aa11848880bb90b8

mpbraendli, 2018-09-18:
The amount of clutter `DecoderAdapter` now prints to `clog` is quite high now. I guess it's just not yet forwarded through the API yet.

You wrote

> @mpbraendli We have to adapt our back-end API to the new behavior.

Did you expect me to do something about that? Who is the "we" in this sentence ? :-)

basicmaster, 2018-09-19:
BTW note that I pushed https://github.com/Opendigitalradio/dablin/commit/16ffe18bd70574239bf98780d785e41e9fc9f4d1 to fix a GCC warning on 18.04.

AlbrechtL, 2018-09-19:
@mpbraendli The we means either you or me. If I do it, it can take some time (currently I'm on family vacation). I you do it, it can be faster and I just have adapt the GUI to the API changes.

@basicmaster I added you to the collaborators list. If you like you can commit the changes yourself.

basicmaster, 2018-09-19:
> @basicmaster I added you to the collaborators list. If you like you can commit the changes yourself.

OK, great - done!

basicmaster, 2018-09-22:
Note that DABlin's FIC decoder not (yet) decodes the service genre (which welle.io currently already displays)...this will be added soon.

mpbraendli, 2018-09-22:
I also prefer the "masking and shifting" approach over the "getbits" approach, for me it's fine to switch. We'll implement the missing features, and we can also take code from ETISnoop.

Do you want to do this before or after the next release? I would prioritise getting a release out above starting a new refactoring task.

mpbraendli, 2018-09-24:
I'll look after the error counters now, and maybe other API changes if useful.

mpbraendli, 2018-09-24:
@basicmaster The information in the format given in `FormatChange()` is a bit redundant with the samplerate and number of channels given in `StartAudio()`.

Could we change the format from `HE-AAC, 48 kHz Stereo @ 64 kBit/s` to `HE-AAC`, removing samplerate, bitrate and "Stereo"?

For DAB, I believe it makes sense to output MPEG version and layer, and mode ("joint stereo" etc.)

mpbraendli, 2018-09-24:
Changed API in 1f08e86. Idea: we should be able to calculate BER before Reed-Solomon more easily now.

AlbrechtL, 2018-09-24:
> Changed API in 1f08e86

Should we move from an integer value to bool within `onFrameErrors` and `onAacErrors`?
Now, the value is all the time one.
https://github.com/AlbrechtL/welle.io/blob/1f08e86c913ae7f76fe860ee5710a3d3e52e12f7/src/backend/decoder_adapter.cpp#L125
 I think also we need some mechanism to reset this value.

> Idea: we should be able to calculate BER before Reed-Solomon more easily now.

Great!

> Do you want to do this before or after the next release? I would prioritise getting a release out above starting a new refactoring task.

Early adopters can use the nightly builds. BTW: I removed the specific next nightly builds package from Bintray to reduce confusion. All nightly builds are uploaded direct into the Bintray master package which is also linked from the welle.io website.

So for me there is no rush to release a version soon. Even inside the GUI there a lot of small bugs and missing features (configurable inputs) that I don't want to see inside the official release. 

What we can do is a alpha release if the Android build is working again. But for me are having backend improvements a higher priority than GUI improvements.

mpbraendli, 2018-09-25:
What do you think about the approach in 6ddf226 ?

Ok about the priorities for the release. I would make the difference between backend improvements and backend issues. Issues and regressions should get fixed, but improvements shouldn't delay the release. But that's just an opinion among many.

AlbrechtL, 2018-09-30:
> What do you think about the approach in 6ddf226 ?

I think the reset has to be come from the dab decoder or dab plus decoder, respectively.
@basicmaster What do you think?

> Issues and regressions should get fixed, but improvements shouldn't delay the release. 

Since you are working on the FIC decoder right now it doesn't make sense to switch it in the next weeks. All changes that you made has to be merged into @basicmaster implementation. You are right we should do this in the next release. I will focus to get GUI working in the next months.

basicmaster, 2018-10-05:
> @basicmaster The information in the format given in `FormatChange()` is a bit redundant with the samplerate and number of channels given in `StartAudio()`.
>
> Could we change the format from `HE-AAC, [...]

I just modified the callback to forward the format info in separate fields; see https://github.com/Opendigitalradio/dablin/commit/d2e8bb44af2a52e9a41ebd1aa9b380e192815f9e. The reason for transmitting some infos again is that e.g. the FDK AAC decoder unfortunately outputs stereo also for mono services (as they in theory could contain PS). So in that case the channel mode of the decoded audio is different to the actual channel mode of the transmission.

basicmaster, 2018-10-05:
> I think the reset has to be come from the dab decoder or dab plus decoder, respectively
> @basicmaster What do you think?

Hmm, I'm not sure what exactly you want to count with `frameErrorCounter`. Currently you count the invalid AUs within one super frame. On which event should a reset (initiated by the audio encoder) occur?

mpbraendli, 2018-10-08:
I also don't know what is best to count.

One very nice metric would be BER before Reed-Solomon. The RS decoder can tell us how many bytes were corrected, and if there were uncorrectable errors. This could be converted to BER, respectively an error flag.

AlbrechtL, 2018-10-08:
Another idea is to count the errors within one second.

AlbrechtL, 2019-12-02:
Closed for now because of no activity.

closed 2019-12-02 by AlbrechtL