AlbrechtL/welle.io #304: [welle-cli] welle-cli detects too many streams

andimik, 2018-09-22:
When I use the BBC DAB stream from 4° East, I discovered that welle-cli adds 7 additional lines, all with service ID 0x4000 (see attached PDF). @mpbraendli seems this is a Swiss ID ;-)

[welle_cli_bbc.pdf](https://github.com/AlbrechtL/welle.io/files/2407590/welle_cli_bbc.pdf)

`ni2http --list` says

```
 0 : BBC Radio 4Extra (0xc22c) Pri subch=12 start=774 CUs= 58 PL=uep 3 bitrate=80
 1 : BBC Radio 1Xtra  (0xc22a) Pri subch=10 start=582 CUs= 96 PL=uep 3 bitrate=128
 2 : BBC AsianNetwork (0xc236) Pri subch= 7 start=486 CUs= 48 PL=uep 3 bitrate=64
 3 : BBC WorldService (0xc238) Pri subch= 9 start=534 CUs= 48 PL=uep 3 bitrate=64
 4 : BBC Radio 1      (0xc221) Pri subch= 1 start=  0 CUs= 96 PL=uep 3 bitrate=128
 5 : BBC Radio 2      (0xc222) Pri subch= 2 start= 96 CUs= 96 PL=uep 3 bitrate=128
 6 : BBC Radio 3      (0xc223) Pri subch= 3 start=192 CUs=140 PL=uep 3 bitrate=192
 7 : BBC Radio 4      (0xc224) Pri subch= 4 start=332 CUs= 96 PL=uep 3 bitrate=128
 8 : BBC Radio 5 Live (0xc225) Pri subch= 5 start=428 CUs= 58 PL=uep 3 bitrate=80
 9 : BBC Radio 6Music (0xc22b) Pri subch=11 start=678 CUs= 96 PL=uep 3 bitrate=128
```

qt-dab detects these streams:

```
BBC Radio 1     ;United Kingdom;C221;1;0;96;128;DAB;UEP 3;3/5;English;Pop Music;
BBC Radio 2     ;United Kingdom;C222;2;96;96;128;DAB;UEP 3;3/5;English;Pop Music;
BBC Radio 3     ;United Kingdom;C223;3;192;140;192;DAB;UEP 3;3/5;English;Arts;
BBC Radio 4     ;United Kingdom;C224;4;332;96;128;DAB;UEP 3;3/5;English;Talk;
BBC Radio 4Extra;United Kingdom;C22C;12;774;58;80;DAB;UEP 3;3/5;English;Talk;
BBC Radio 5 Live;United Kingdom;C225;5;428;58;80;DAB;UEP 3;3/5;English;News;
BBC Radio 6Music;United Kingdom;C22B;11;678;96;128;DAB;UEP 3;3/5;English;Rock Music;
BBC Radio 1Xtra ;United Kingdom;C22A;10;582;96;128;DAB;UEP 3;3/5;English;Other Music;
BBC AsianNetwork;United Kingdom;C236;7;486;48;64;DAB;UEP 3;3/5;English;Talk;
BBC WorldService;United Kingdom;C238;9;534;48;64;DAB;UEP 3;3/5;English;Talk;

BBC Guide       ;United Kingdom;E1C79E5E;62;832;24;32;0;EEP 3-A;0;no;;
```

There is no 4000 in the whole stream:

`etisnoop -i bbc.eti | grep 4000`

[bbc.eti.zip](https://github.com/AlbrechtL/welle.io/files/2407589/bbc.eti.zip)

k-otixxx, 2018-10-07:
Is welle-cli available under Linux?
And which compiler you used?

mpbraendli, 2018-10-09:
In the `fib-processor`, services are created in `findServiceId()`, which gets called from many different FIGs. This is not exactly ideal, it would be better to create services only when FIG0/2 gets parsed.

BTW, I don't get an invalid service 0x4000 with your ETI file, maybe because we're not doing the additional processing step between ETI and welle-io in the same way, however I get a service with `0xc232` from FIG0/16 (other ensemble programme number).

I'll continue investigation.

mpbraendli, 2018-10-09:
I've changed the way the service list gets filled. Can you try again?

andimik, 2018-10-09:
Confirmed, the "ghost" service IDs disappeared.

BTW: I would comment line 295 in `index.js`, so that the text

> DAB, 48000 Hz, MPEG 1.0 Layer II, 48 kHz Joint Stereo

gets 

> DAB, MPEG 1.0 Layer II, 48 kHz Joint Stereo

as we have the 48 kHz twice ...

closed 2018-10-09 by andimik

andimik, 2018-10-09:
BTW: What does welle-io mean with

- PL UEP 35
- PL UEP 45
- PL UEP 21 and
- PL UEP 16 in our case here?

As you can see in https://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=G&inputEn=CE15 all services use the same protection level UEP 3.

Seems this is calculated from the used CUs.

I have not found a spec or doc which tells me what UEP 35 should be. When I google for it, I only find my PDF from this issue ...

mpbraendli, 2018-10-10:
Thanks for confirming it works.

On the 480kHz being written twice: is being discussed here https://github.com/AlbrechtL/welle.io/issues/302#issuecomment-423912831

I agree the table index printed for UEP is not very meaningful. Tracked here https://github.com/AlbrechtL/welle.io/projects/1#card-13731805

andimik, 2018-10-10:
Ok thanks, the 2nd link is not accessible for me.