AlbrechtL/welle.io #306: [Android Auto] [welle-cli] [Feature request] Remote control welle-cli from Android App

rubenverhoef, 2018-09-25:
I want to use welle.io in my car using my diy project which include a rpi and crankshaft (using openauto).

We have two options here:
1. Do the decoding at the phone with RTL_TCP (not yet supported) 
2. Do the decoding at the RPI with welle-cli

I want to use option two because when using AA wireless option one will result in a big data rate on the wifi signal. But for option two i need to have the ability to (remote) control the welle-cli server from the android app (welle.io) on my phone. AA support for the android app is already there so it is only the remote control part what is missing for this situation. So this is a Feature request for this matter.

Issues which can be fixed with this:
#239 #187 and #224

mpbraendli, 2018-09-25:
What remote control feature are you requiring?

welle-cli presents a API over HTTP, so you can design an android app or even use the web browser to connect to the RPi over WiFi and control welle-cli.

The welle-io android app doesn't use the HTTP API welle-cli offers, and is therefore difficult to adapt to it. If I understand your feature request like that, that is, "adapt the welle.io android app so that it can remote-control a welle-cli instance", then I feel it'll be unlikely we will consider it. It's probably a better approach to design a nice webpage or app dedicated to this purpose. I doubt this will be a priority for us, I hope you understand.

Did I interpret your request correctly?

rubenverhoef, 2018-09-25:
You are correct. The problem of making a simple app is that it has to support android auto as well. So i was thinking include it in the welle.io android app would also makes so much possible, like a RPI in the living room with welle.io (CLI) and control it with the android app.

mpbraendli, 2018-09-25:
I'll let @AlbrechtL take the decision if he wants to take this into a TODO list or not.

AlbrechtL, 2018-09-30:
@rubenverhoef welle.io has only enabled Android Auto (AA) inside the APKs at Github. The version inside the Play Store doesn't have AA because Google denied it. We don't know why Google doesn't allow welle.io with AA inside the Play Store (we fulfilled all requirements). But one reason could be that Google couldn't test welle.io in the US (the USA doens't have DAB broadcasts). 

Furthermore the welle.io Android version is an all in solution. Your use case is complete different from the origin of welle.io. @mpbraendli is right that the correct approach for your use case is to implement an app which consumes and controls the welle-cli stream and API.

closed 2018-09-30 by AlbrechtL