AlbrechtL/welle.io #307: cross compiling for Raspberry Pi

Vitalic66, 2018-10-07:
Tried to follow instructions for RPi cross compiling but when i try to start welle-io on Pi, i get the error, that the binary file can't be opened.

I also have problems while testing the qopenwidget. I followed the insturcion and "removed" EGL/GLES libraries and added symbolic links, but when i boot to command line, qopenwidget outputs the error about missing libraries.

Is it also possible to run welle-io on x desktop of Pi?

mpbraendli, 2018-10-08:
It is possible to run welle-io on Raspberry Pi, but without further details from you about the compilation errors you get, there's no way to offer you help.

In general, I'd advise against cross-compiling, it's a hassle to set up.

MPunktBPunkt, 2018-10-11:
Hi is there a ready image of welle.io for the raspberry to download somewhere?

Vitalic66, 2018-10-17:
Finally managed to build welle.io on Raspberry Pi 3. I did not crosscompile it. I compiled Qt 5.10 on the Pi (also not crosscompiled) and build welle.io on Pi.

Took several hours and many tries to find the right settings but now i got a running welle.io on the pi.

I use a RTL_SDR usb dongle and a FM/DAB splitter in my car.

https://youtu.be/IhriZQVgTwY

mpbraendli, 2018-10-19:
No Qt will stay a dependency. However the receiver backend is not Qt-less and there's a command-line frontend that is also compilable without Qt. But if you want the GUI, you need Qt.

Is there no way to use the packaged Qt to compile welle-io? That would simplify the process a lot.

Vitalic66, 2018-10-19:
I guess 5.7.1 is too old with the latest version of welle.io. 
Yes it took several hours to compile 5.10.1 but make -j4 or -j2 caused errors after compiling.
When will the new version be available? Thinking about a custom layout for my raspi car project.

mpbraendli, 2018-10-22:
Yeah, it seems that only debian testing offers a recent enough Qt according to
https://packages.debian.org/qt5-default

closed 2018-10-25 by AlbrechtL