AlbrechtL/welle.io #308: first frequencies scanned to high?..

filipdns, 2018-10-07:
Hello,

I try to use your binary but it's starting in bloc 5A at 174Mhz but here the first radio frequency start at 88Mhz, how can I receiver my local FM radio?

kind regards
Philippe

mpbraendli, 2018-10-08:
Hi @filipdns 

welle-io is a DAB receiver, not a FM receiver. Maybe in the future it'll be a FM receiver too, but somebody needs to do that.

Best regards

closed 2018-10-08 by mpbraendli

filipdns, 2018-10-08:
I try many one but often they are made for mingw not visual studio and lot of dependencies are different.

My question was to have FM with this because the design is nice but after scanning (I see on spectrum analyser that a signal is received), nothing is found and I'm supposed to have some radio available near of me...

I try SDR-Sharp for Windows and it's working fine, then the problem is not from my sdr usb key.

filipdns, 2018-10-09:
Hello andimik
you are not completely wrong, DAB is new for me and I'm referring only on what I read recently. My country provide DAB+ and the nearest station is at around 50 km from me, then, I may be wrong to assume that I should receive these stations from my home.
I don't disturb this project for my questions, sorry for that.
Kind regards
Philippe