AlbrechtL/welle.io #309: Plots (Tiles) legend suggestions

stanroehrich, 2018-10-08:
Constellation Diagram:
Horizontal legend 'Subcarrier' is good but the scale number from 0 to 1200 is not corresponding to the real sub-carrier numbers. 

Impulse response Diagram:
Horizontal legend 'Frequency' is wrong. It can be  'samples /us/km'  (the ability to display microseconds or km is useful for such view)

notes: very nice tool ;)  /Best

AlbrechtL, 2018-10-08:
Please post updated figures how it should be.

mpbraendli, 2018-10-15:
CIR axis changed to "Samples" in c31208d

mpbraendli, 2018-10-26:
It should be from -768 to 768 (that's the parameter `k` you also find in the spec)

But I've noticed that the x-values of the plot points do not correspond to `k` in any meaningful way. I will debug this.

mpbraendli, 2018-10-31:
Constellation plot is improved a bit in 672abe6 but I have the impression there's still something wrong about the index calculation.

mpbraendli, 2018-10-31:
@AlbrechtL is it easy to do a tooltip on the CIR plot that shows microseconds and km-equivalent of the x value (now `samples`), one that follows the mouse cursor and gets updated as you move the mouse? Is it possible to do that entirely in the QML?

Conversion is:
* `microseconds` = `samples` / 2.048
* `km` = `microseconds` × 3e5

AlbrechtL, 2018-11-01:
I added a something in dc888e2f7daa6c7ea512de818cb7e2fe5198f166.

Dou you mean this?
[image]

mpbraendli, 2018-11-01:
Oh! Exactly! Awesome! Thanks a lot!

I will probably modify it s.t. 512 becomes zero, because that's where we actually place the FFT window.

AlbrechtL, 2019-03-03:
I reintroduced this feature in https://github.com/AlbrechtL/welle.io/commit/f42c18c63ecb9291b2dd4f0ac8c3b8851037d6b2.

closed 2019-03-03 by AlbrechtL