AlbrechtL/welle.io #310: newDebugOutput() used after GUI deleted

gvanem, 2018-10-23:
Compiling using Qt5 and MSVC in debug-mode (`-MDd`) shows that welle-io is calling `CGUIHelper::newDebugOutput()` after `delete guiHelper;` was called.  But a freed object in debug-mode is typically filled with a series of `0xDD`. Hence the crash on this instruction:
```
mov     ecx,dword ptr [eax+10h] ds:002b:dddddded=????????
```
Where `eax=dddddddd`. For reference, here is some of the call-stack.
```
Qt5Cored!QMetaObject::activate(class QObject * sender = 0x09f68be8, int signalOffset = 0n3, int local_signal_index = 0n6, void ** argv = 0x001df388)+0x3a [c:\users\qt\work\qt\qtbase\src\corelib\kernel\qobject.cpp @ 3643] 
Qt5Cored!QMetaObject::activate(class QObject * sender = 0x09f68be8, struct QMetaObject * m = 0x0055f3c4, int local_signal_index = 0n6, void ** argv = 0x001df388)+0x21 [c:\users\qt\work\qt\qtbase\src\corelib\kernel\qobject.cpp @ 3633] 
WELLE_IO!CGUIHelper::newDebugOutput(class QString _t1 = class QString)+0x47 [f:\gv\dx-radio\welle-io\msvc_obj\moc_gui_helper.cpp @ 417] 
WELLE_IO!CGUIHelper::setNewDebugOutput(class QString text = class QString)+0x60 [f:\gv\dx-radio\welle-io\src\welle-gui\gui_helper.cpp @ 504] 
WELLE_IO!CDebugOutput::handleMessage(class QString str = class QString)+0x13c [f:\gv\dx-radio\welle-io\src\welle-gui\debug_output.cpp @ 86] 
WELLE_IO!CDebugOutput::customMessageHandler(QtMsgType type = QtDebugMsg (0n0), class QMessageLogContext * __formal = 0x0d512208, class QString * str = 0x0d5121f8)+0x2c3 [f:\gv\dx-radio\welle-io\src\welle-gui\debug_output.cpp @ 58] 
Qt5Cored!qt_message_print(QtMsgType msgType = QtDebugMsg (0n0), class QMessageLogContext * context = 0x0d512208, class QString * message = 0x0d5121f8)+0xab [c:\users\qt\work\qt\qtbase\src\corelib\global\qlogging.cpp @ 1793] 
Qt5Cored!qt_message_output(QtMsgType msgType = QtDebugMsg (0n0), class QMessageLogContext * context = 0x0d512208, class QString * message = 0x0d5121f8)+0x14 [c:\users\qt\work\qt\qtbase\src\corelib\global\qlogging.cpp @ 1851] 
Qt5Cored!QDebug::~QDebug(void)+0xc5 [c:\users\qt\work\qt\qtbase\src\corelib\io\qdebug.cpp @ 154] 
WELLE_IO!main(int argc = 0n5, char ** argv = 0x0085dea8)+0x2139 [f:\gv\dx-radio\welle-io\src\welle-gui\main.cpp @ 380] 
```
The last line is right after `delete guiHelper;`.

Seems Qt5 allows one to shoot yourself in the foot. But moving the `qDebug()` before the `delete`, fixes the crash:
```diff
--- a/src/welle-gui/main.cpp 2018-10-15 18:33:47
+++ b/src/welle-gui/main.cpp 2018-10-23 13:08:32
@@ -287,11 +287,11 @@
     // Run application
     app.exec();

+    qDebug() << "main:" <<  "Application closing";
+
     // Delete the RadioController controller to ensure a save shutdown
     delete guiHelper;
     delete radioController;

-    qDebug() << "main:" <<  "Application closed";
-
     return 0;
```

AlbrechtL, 2018-10-25:
Fixed in c9d148c7ef9475d3b50bc59e483aff90a85bf2a1. Thanks!

Please open a pull request next time.

closed 2018-10-25 by AlbrechtL

gvanem, 2018-10-25:
> Please open a pull request next time.

I'm still a Git neophyte. Besides there are more problems with Debug-mode. You should try it.