AlbrechtL/welle.io #311: Compile on macOS Mojave (10.14) gives black screen

vk1kcm, 2018-10-25:
The app will compile on macOS Mojave using QT5.9 and seems to run, however the interface only shows a black screen.  Clicking in the interface where the station should be works, and the station starts to play so the app is working, it's just nothing is displayed.

This seems to be a common issue with OpenGL/GLUT and QT5 QML applications on Mojave and seems to be a bug in the 10.14 OpenGL frameworks in Mojave.  However it does make the application useless.

There is an bugreport in the QT issues register saying that it won't be fixed in QT5.9 as Xcode 10 is unsupported.  
https://bugreports.qt.io/browse/QTBUG-71206

Workaround seems to be to compile under macOS High Sierra, or using the 10.13 toolchain.  I'm still working out how to do that.

AlbrechtL, 2018-10-25:
You can try to comment in the following line to use software OpenGL
https://github.com/AlbrechtL/welle.io/blob/c9d148c7ef9475d3b50bc59e483aff90a85bf2a1/src/welle-gui/main.cpp#L66

vk1kcm, 2018-10-27:
Hi Albrecht, I was using the main branch and that line isn't there.  I've grabbed the "next" branch and installed Q5 5.11 and tried it again and it works fine.  I think it's only an issue with Qt5 5.9.

AlbrechtL, 2018-11-01:
Thanks for the feedback and great that it is working!

closed 2018-11-01 by AlbrechtL