AlbrechtL/welle.io #312: A few regressions since 52aa420

mpbraendli, 2018-10-31:
A couple of things I noted since commit 52aa420:

* [ ] The command line option `-d` has no effect anymore. It would ideally override the last used setting done in the GUI
* [ ] Cannot switch back from file input to RTLSDR. It now crashes with a segfault calling `input.restart();` in `void OFDMProcessor::start()`. I've recovered it by checking the auto input box, closing welle.io and opening it again.

I don't know if you had plans about fixing these two things already, and it's just in progress for now.

AlbrechtL, 2018-11-01:
You are correct about these. It's WIP.
Step for step I add more GUI elements to get rid of the command line options. The main motivation is that Android doesn't have a CLI and it's a hassle under Windows.

mpbraendli, 2018-11-01:
Ok, I understand.

I'll let you decide if you want to keep the issue open or if you want to track this elsewhere.

AlbrechtL, 2018-11-03:
The rtl-sdr issue should be fixed now in 9dfdf7fbb5a71941a8f4c3aba2b0a36027b09d15.