AlbrechtL/welle.io #313: No sound on  20181031_0321398_Linux_welle-io-x86_64.AppImage

erikdsgh, 2018-11-01:
Running on Ubuntu 16.04 LTS.
No sound.
Graphics show signal as usually.
Previous version  20181001_5dcf0e7_Linux_welle-io-x86_64.AppImage does not have that problem.
If more info is needed please send me instructions on how to obtain it.

AlbrechtL, 2018-11-01:
That's interesting because we didn't change the sound interface.

erikdsgh, 2018-11-01:
If more info can help you, please tell
      me what i should provide and how to obtain it.
      
      Best regards.
      
      Erik.
      ------------------------------------------
      
      Albrecht Lohofener wrote:
    
    
      That's interesting because we didn't change the sound
        interface.
      —
          it on GitHub, or mute
          the thread.
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/AlbrechtL/welle.io/issues/313#issuecomment-435101102",
"url": "https://github.com/AlbrechtL/welle.io/issues/313#issuecomment-435101102",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [AlbrechtL/welle.io] [Linux] No sound on  20181031_0321398_Linux_welle-io-x86_64.AppImage (#313)",
"sections": [
{
"text": "",
"activityTitle": "**Albrecht Lohofener**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@AlbrechtL",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"AlbrechtL/welle.io\",\n\"issueId\": 313,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"AlbrechtL/welle.io\",\n\"issueId\": 313\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/AlbrechtL/welle.io/issues/313#issuecomment-435101102"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 404281096\n}"
}
],
"themeColor": "26292E"
}
]

erikdsgh, 2018-11-01:
[image]

erikdsgh, 2018-11-01:
Same problem with  20181101_dc888e2_Linux_welle-io-x86_64.AppImage

erikdsgh, 2018-11-01:
Found the issue!
In the last 2 versions the backend section now has 'out of the box' the parameter coarse corrector disabled. This parameter is now called "Enable coarse corrector"

In the versions before the last 2 the parameter was "Disable coarse corrector" and was 'out of the box' not set.

Sorry for all the trouble.

closed 2018-11-01 by erikdsgh

AlbrechtL, 2018-11-03:
Yes, I enabled the coarse corrector by default. Great that welle.io is working again for you!