AlbrechtL/welle.io #314: "CFaadDecoder: warning: No standard extension payload allowed in DRM" Segmentation fault

ROV28, 2018-11-03:
I got the following warning:
```
[X-PAD len] CFaadDecoder: can't write WAV sample to buffer. Wrote 0 bytes of 3840 bytes.
CFaadDecoder: can't write WAV sample to buffer. Wrote 0 bytes of 3840 bytes.
CFaadDecoder: warning: No standard extension payload allowed in DRM
CFaadDecoder: can't write WAV sample to buffer. Wrote 0 bytes of 3840 bytes.
CFaadDecoder: can't write WAV sample to buffer. Wrote 0 bytes of 3840 bytes.
CFaadDecoder: warning: No standard extension payload allowed in DRM
CFaadDecoder: can't write WAV sample to buffer. Wrote 0 bytes of 3840 bytes.
CFaadDecoder: can't write WAV sample to buffer. Wrote 0 bytes of 3840 bytes.
CFaadDecoder: warning: No standard extension payload allowed in DRM
Segmentation fault 
```

Platform is Fedora 28 Linux 
Welle.io: Latest commit 0141eb1 on 30 Aug

AlbrechtL, 2018-11-03:
Duplicated, see here (https://github.com/AlbrechtL/welle.io/issues/30).

closed 2018-11-03 by AlbrechtL

ROV28, 2018-11-04:
Not exactly a duplicate, an older version of Welle.io had identical symptoms as described at at issue #30 , the latest version has no breaking audio but crashes.

AlbrechtL, 2018-11-06:
The origin of the issues are still the same I suppose. The main issue is that I can't reproduce it on my system. There is something wrong with the libfaad.

ROV28, 2018-11-07:
That could be, I am not sure.

For information these are the the "faad" related packages on my system:
- faad2-2.8.8-1.fc28.x86_64
- faad2-libs-2.8.8-1.fc28.x86_64
- faad2-devel-2.8.8-1.fc28.x86_64

Are these newer versions than the one you are using?

AlbrechtL, 2018-11-09:
As I wrote in https://github.com/AlbrechtL/welle.io/issues/30 I have no idea what the real problem is.

AlbrechtL, 2018-11-10:
Great! But nevertheless it is strange. As far as I know there are no DRM data inside the AAC stream.

andimik, 2018-12-16:
BTW: I had the same problem, I have compiled faad with DRM (because I wanted to play with DRMplus), and got a very strange sound and lots of these warning.

Seems I really have to disable it ...

andimik, 2019-01-04:
The hint on https://github.com/AlbrechtL/welle.io/issues/314#issuecomment-437570777 did NOT help.