AlbrechtL/welle.io #315: Compil error from 9dfdf7

YoannQueret, 2018-11-06:
From commit 9dfdf7fbb5a71941a8f4c3aba2b0a36027b09d15, welle.io CLI can not compil. (Debian 9.4 on ARM)

```
[ 11%] Building CXX object CMakeFiles/welle-cli.dir/src/welle-cli/tests.cpp.o
In file included from /opt/welle.io/src/backend/radio-receiver.h:38:0,
                 from /opt/welle.io/src/welle-cli/webradiointerface.h:36,
                 from /opt/welle.io/src/welle-cli/welle-cli.cpp:50:
/opt/welle.io/src/backend/radio-controller.h:40:15: fatal error: any: No such file or directory
 #include <any>
               ^
compilation terminated.

CMakeFiles/welle-cli.dir/build.make:110: recipe for target 'CMakeFiles/welle-cli.dir/src/welle-cli/webradiointerface.cpp.o' failed
make[2]: *** [CMakeFiles/welle-cli.dir/src/welle-cli/webradiointerface.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/welle-cli.dir/all' failed
make[1]: *** [CMakeFiles/welle-cli.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

andimik, 2018-11-06:
Same problem here. 

I even have updated to the cmake 3.13.0

AlbrechtL, 2018-11-06:
You need to update your compiler because I'm using some C++17 features. With GCC 7.0 it should work.

closed 2018-11-06 by AlbrechtL

andimik, 2018-11-06:
Update: It builds, but I get an error:

> ./welle-cli: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.11' not found (required by ./welle-cli)

jay-to-the-dee, 2018-11-07:
Can confirm this "issue" on Ubuntu 18.04. However quite easy to fix. Simply install package `g++-8` 
 and add the following towards the top of CMakeLists.txt:
```
# THIS HAS TO COME BEFORE THE PROJECT LINE
set(CMAKE_C_COMPILER "gcc-8")
set(CMAKE_CXX_COMPILER "g++-8")
```

Thanks to @AlbrechtL for being clear as to the changes that are causing this behaviour!

mpbraendli, 2018-11-07:
It's really painful for me to require C++17 on the machines where I deploy welle-cli. On the dev machine it's fine, but the production systems are not bleeding-edge enough for this. Is it ok for you that I replace your usage of `std::any` with something C++14?

mpbraendli, 2018-11-07:
Can you have a look at 09d9758 and tell me what you think about that approach?

1. Use string-parameters between GUI and RadioController.
1. Use a device-agnostic enum to represent parameters for the devices.

If one day we need something that isn't representable as `int` we can use strings instead, or move to an approach where each possible setting is a class inheriting from a common base class, and contains both the information about the param (the class), and the value to set (field inside the class). Device functions would then take a `shared_ptr<base_class>` and use `dynamic_cast` to get the value out.

gvanem, 2018-11-07:
> It's really painful for me to require C++17 on the machines where I deploy welle-cli. 

I just discovered this `<any>` horrid stuff. It surely breaks in MSVC / clang-cl.

AlbrechtL, 2018-11-07:
@mpbraendli Your changes are OK. As longs as the functionality is given everything is fine.

mpbraendli, 2018-11-07:
@AlbrechtL thanks, I hope I didn't break anything, have no way of testing this.

AlbrechtL, 2018-11-09:
We will see it ;-)