AlbrechtL/welle.io #316: Mono stations only play out of one channel

jay-to-the-dee, 2018-11-07:
This is a low priority feature enhancement more than anything.

Unfortunately in the UK a lot of commercial DAB stations are not only using the old inefficient MP2 codec, but are also broadcast in Mono only. When selecting such a station, the audio comes out of only one of my speakers which makes it sound much quieter than the full stereo stations. Is it possible to duplicate the mono audio stream across both left and right channels to remove this effect?

Cheers,

AlbrechtL, 2018-11-07:
Interesting. This mono on both speakers was working in previous versions. Which version did you use?

andimik, 2018-11-07:
Yes, confirmed. Mono plays on left channel only. Using latest commit.

mpbraendli, 2018-11-07:
@basicmaster how is dablin handling DAB-mono? Does it upconvert it to stereo internally?

Before commit a8391cb we had code in `Mp2Processor::PutAudio` that converted mono to stereo, maybe that changed.

basicmaster, 2018-11-07:
In case of MP2, the decoded stream is forwarded without channel count conversion i.e. a mono MP2 station results in a single audio channel. So in the mono MP2 station case, the `StartAudio` callback is called with `channels` set to 1.

Note that `channels` here does not necessarilly correspond to the actual channel count, as the (alternate) FDK AAC decoder unfortunately outputs stereo for mono channels, too. The reason is that this decoder assumes that PS could be present...

AlbrechtL, 2018-11-09:
I think we have to put new conversation code to `void DecoderAdapter::PutAudio(const uint8_t *data, size_t len)` to do the conversation to two channels.
https://github.com/AlbrechtL/welle.io/blob/6ddf226b8efadb9a30e1585ebd07b66f8aa08108/src/backend/decoder_adapter.cpp#L92-L118

AlbrechtL, 2018-11-11:
@basicmaster Thanks for fixing!

closed 2018-11-11 by AlbrechtL

jay-to-the-dee, 2018-11-17:
Can confirmed fixed. Thanks everyone!