AlbrechtL/welle.io #317: Software Crash at Startup

nousername1, 2018-11-08:
I use the normal SDR Dongle who work with SDR#.
Welle.IO crash when I startup. Maybe the same error someone sayed before with the Chartset or how does it called.
Where is the Channel list stored?
My OS is W7 and it run on a current Thinkpad.

gvanem, 2018-11-08:
> Welle.IO crash when I startup.

Details please. Maybe a stack-dump?

> Where is the Channel list stored?

At `HKEY_CURRENT_USER\Software\welle.io\welle.io\stationListSerialize`. A Json `REG_SZ` string.

nousername1, 2018-11-08:
`Problemsignatur:
  Problemereignisname:	APPCRASH
  Anwendungsname:	welle-io.exe
  Anwendungsversion:	0.0.0.0
  Anwendungszeitstempel:	5a6635e5
  Fehlermodulname:	ig9icd32.dll
  Fehlermodulversion:	21.20.16.4727
  Fehlermodulzeitstempel:	59565458
  Ausnahmecode:	c0000005
  Ausnahmeoffset:	004ea3a6
  Betriebsystemversion:	6.1.7601.2.1.0.256.48
  Gebietsschema-ID:	3079
  Zusatzinformation 1:	0a9e
  Zusatzinformation 2:	0a9e372d3b4ad19135b953a78882e789
  Zusatzinformation 3:	0a9e
  Zusatzinformation 4:	0a9e372d3b4ad19135b953a78882e789
`

gvanem, 2018-11-08:
That was very useful, not!

Except the `Fehlermodulname: ig9icd32.dll` hints on a problem with the Intel graphics driver.

AlbrechtL, 2018-11-09:
@gvanem Don't be rude with nousername1! He reported a problem and he is helping to solve it!

@nousername1 Which version did you use?

AlbrechtL, 2018-12-08:
@nousername1 Any process? Otherwise I will close this issue.

nousername1, 2018-12-08:
It still not work.

AlbrechtL, 2018-12-08:
Which version do you use?

AlbrechtL, 2018-12-27:
Closed because of no activity.