AlbrechtL/welle.io #318: Incorrect #include

russel, 2018-11-08:
It seems that src/welle-gui/gui_helper.cpp includes src/input/airspy_sdr.h which inscludes libairspy/airspy.h even when -DAIRSPY=TRUE was *not* set during CMake build creation.
```
In file included from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-gui/gui_helper.cpp:41:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/input/airspy_sdr.h:45:10: fatal error: libairspy/airspy.h: No such file or directory
 #include "libairspy/airspy.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

gvanem, 2018-11-08:
I reported this yesterday: https://github.com/AlbrechtL/welle.io/commit/9dfdf7fbb5a71941a8f4c3aba2b0a36027b09d15#diff-53282073a4f81199e8ca97257db93175

Wait until someone fixes it. Or patch by hand.

AlbrechtL, 2018-11-09:
@gvanem Please open an pull request.

mpbraendli, 2018-11-10:
Hey all,
How's commit e8e487e working for you?

russel, 2018-11-10:
Branch next now builds for me. I have configured to use only SoapySDR.

AlbrechtL, 2018-11-11:
@mpbraendli Thanks for fixing!

closed 2018-11-11 by AlbrechtL