AlbrechtL/welle.io #319: Does not run on Clear Linux OS

probonopd, 2018-11-10:
Does not run on Clear Linux OS:

```
      1175:	/usr/lib64/libfontconfig.so.1: error: symbol lookup error: undefined symbol: FT_Done_MM_Var (fatal)
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Available platform plugins are: xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)
```

Not bundling FreeType solves this.

AlbrechtL, 2018-11-11:
I don't really understand the problem.
> Not bundling FreeType solves this.

What do you mean with this?

AlbrechtL, 2018-11-11:
Thanks for reporting this issue. Can you give me a hint what I can do?

I mean you created the initial [.travis.yml](https://github.com/AlbrechtL/welle.io/blob/571532eae289cba8927b8c0442763bc4fecc4a37/.travis.yml).

probonopd, 2018-11-11:
Probably just running a new build would be sufficient, since the AppImage tools have the FreeType library blacklisted nowadays.

AlbrechtL, 2018-11-11:
When did you blacklist the FreeType library?

Can you test the latest nightly build? https://bintray.com/albrechtl/welle.io/welle.io_nightly#files

probonopd, 2018-11-11:
> When did you blacklist the FreeType library?

8 months ago, https://github.com/AppImage/AppImages/commit/9082bbe3b924b656d8675da64a69c6b2b32ac7b4

AlbrechtL, 2018-11-11:
The device should work anyway. The error message is a bug.

probonopd, 2018-11-11:
Indeed! When launching the AppImage linked above as root, I do get a read error, but nevertheless I can search for stations and listen to them. So, working for me (more or less :-))

[image]