AlbrechtL/welle.io #320: Fails to use RTL-SDR device

TheAssassin, 2018-11-11:
welle.io is unable to use my RTL-SDR receiver.

[image]

```
RTL_SDR: Open rtl-sdr
RTL_SDR: Found 1 devices. Uses the first working one

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
RTL_SDR: Opening rtl-sdr failed
```

The device works fine with other applications like gqrx.

I'm using the AppImage.

AlbrechtL, 2018-11-11:
Please try the latest [nightly build](https://bintray.com/albrechtl/welle.io/welle.io_nightly#files) please.

AlbrechtL, 2018-11-11:
Duplicated, see here https://github.com/AlbrechtL/welle.io/issues/178

TheAssassin, 2018-11-11:
With the nightly, I get the same error message as in #319. The device seems to work. When restoring the application from tray, the app crashed, though, and I had to plug the SDR device out and in again in order to make it work again.

probonopd, 2018-11-11:
Try to run the AppImage as root, then search for stations. Worked for me indeed! (Despite the red error.)

AlbrechtL, 2018-11-11:
The error message is a bug. It will be fixed in the near future.

closed 2018-11-11 by AlbrechtL