AlbrechtL/welle.io #322: DAB mode II not working

AlbrechtL, 2018-12-08:
Ongoing discussion from https://github.com/AlbrechtL/welle.io/commit/8f7a5cf0428caacfb8a3033210d34950c0f4eea8#commitcomment-31543899

mpbraendli, 2018-12-10:
I've quickly checked with a rawfile containing Transmission Mode II. The receiver itself works, but there is no way to set it to mode II except changing the `setMode(1)` line 298 in `dab-constants.cpp` to `setMode(2)`. The `RadioReceiver` doesn't offer an argument in the constructor to set this to another value. I added an argument in f4b721a.

The `CRadioController` has no awareness of mode. How do you want to represent this?

1. auto mode (L-Band implies Mode 2, otherwise Mode 1) + an override in the settings. Keep in mind RAW file input.
1. manual setting, user must choose Mode (how to get scan to work? Skip L-Band in scan and document "if you want to listen to L-Band DAB, you must find out channel and mode yourself"?)

andimik, 2018-12-10:
Yes, if you have live reception the L-Band can be used for Mode 2.

But (as a normal user) raw files are not easy to find out why they fail.

andimik, 2018-12-10:
BTW: For qt-dab-2.0 Jan uses an entry in the `qt-dab.ini` config-file to set to Mode 2 for raw-files (not tested, but he told me that).

AlbrechtL, 2018-12-10:
> 1. auto mode (L-Band implies Mode 2, otherwise Mode 1) + an override in the settings. 

I really like this idea. A regular user shouldn't know the Mode.

> Keep in mind RAW file input.

We can simply add an option to the RAW file settings. Like the encoding.

> 2. manual setting, user must choose Mode

I don't think we need this. 

Does somebody know if hams also have special DAB modes?

mpbraendli, 2018-12-11:
I'm not aware of any ham usage of DAB.

AlbrechtL, 2019-03-03:
Can someone provide a RAW file with DAB mode II?

mpbraendli, 2019-03-04:
There you go: http://mpb.li/pub/20190304_bbc.modeII.u8.iq.gz

andimik, 2019-03-04:
Or see https://www.rundfunkforum.de/viewtopic.php?f=11&t=56477&p=1455354

AlbrechtL, 2019-03-06:
Thanks for the files.

I improved the DAB params handling in  63592df994a539ee9dfe10199ae288e4b78842b9

Now it is possible more easily to change the DAB mode inside the code
https://github.com/AlbrechtL/welle.io/blob/63592df994a539ee9dfe10199ae288e4b78842b9/src/welle-gui/radio_controller.cpp#L286

 And to test DAB mode 2.
[image]

1. Are they any ideas to detect to DAB mode automatically?
2. It seems that the FIC CRC flag is not working properly

mpbraendli, 2019-03-07:
1. Thinking out loud: you could check for the null symbol in 24ms intervals, and see a different pattern depending on the transmission mode. TM 1 has 96ms frames, TM 2 and 3 24ms, TM 4 48ms.

On the other hand, the latest DAB specification has removed TM 2, 3 and 4, and we can expect that future DAB transmissions will only be TM 1. We could argue that a manual setting suffices for the uncommon cases.

2. I'll write that down in my TODO list.

mpbraendli, 2019-04-23:
I checked the FIC CRC indicator flag, and it looks fine here. (Changes in 54f66c9 are not directly related to this)

AlbrechtL, 2019-04-25:
> Thinking out loud: you could check for the null symbol in 24ms intervals, and see a different pattern depending on the transmission mode. TM 1 has 96ms frames, TM 2 and 3 24ms, TM 4 48ms

Is it worth to try this approach to detect the mode?

mpbraendli, 2019-04-29:
It could work. Whether it is worth trying it out is not really a technical question, but rather a question about the desire to have this feature or not. That's up to you!

I believe auto (Band III → TM 1; Band L → TM 2) with an override in the settings is the way to go. It's simpler to implement and fits most needs.

AlbrechtL, 2019-04-30:
> Whether it is worth trying it out is not really a technical question, but rather a question about the desire to have this feature or not. That's up to you!

For me personally I don't need the band L because at my location (Germany) there are only band III transmissions.
On the other hand welle.io is a open source application and the FM list guys may need this feature. 

How may users are needing TM 2  - I don't know. Even in version 1.0 TM 2 was only available with a command line option. Nobody complained about it or complained about non existing stations. So I think  that TM 2 has not so much users. This was also the reason why I removed the command line option. Now I know that at least one or two users are really using it.

So if you like you can implement it.

mpbraendli, 2019-05-02:
Okay, so I undersand it would be a nice to have feature.

Allow me to be transparent: I won't work on that however, I have enough things to do already.

We could flag the issue as being accessible for new people joining the project, maybe tag it "help wanted".

AlbrechtL, 2019-05-03:
Done.

AlbrechtL, 2024-10-08:
Closed because of inactivity.