AlbrechtL/welle.io #334: welle.io_2.0-Beta1_Linux_x86_64.AppImage won't exit correctly

flux242, 2019-01-03:
when gui is closed apprun would continue to work

AlbrechtL, 2019-01-07:
Thanks for reporting!

What do you mean with "apprun"?

flux242, 2019-01-13:
when the main window is closed the application keeps running. To close it one would need to interact with a small icon. Why is that? Why can't one just close it by closing the main window? AppRun is the main process that is started in an appimage container. Check AppImage for details

AlbrechtL, 2019-01-13:
What do you mean with "small icon"?
Please try it again with the latest nightly (I fixed a crash bug while closing).

flux242, 2019-01-13:
"small icon" is an image with limited pixel resolution along both axis. "Small" in this context means that it covers very small portion of the whole visible display area. Please make it possible to terminate the process without the need to interact with a small systray icon

AlbrechtL, 2019-01-16:
I don't quite understand the issue. If you close the application by the X then it stops running. If you minimize the the program then you have to deal with the icon.

flux242, 2019-01-19:
well if started under xfce de then it works as you describe above. But I'm using i3 window manager and it doesn't work as described. If I close the window then it minimizes itself instead of terminating the process. It worked correctly before 2.0 (where I believe the systray icon was added)

AlbrechtL, 2019-01-19:
It sounds like an issue with i3. I suggest to ask the i3 developers what's going wrong. Under KDE and GNOME it is working correctly.

closed 2019-01-19 by AlbrechtL