AlbrechtL/welle.io #336: ringbuffer is using deprecated API on macOS 10.12

mattes-bru, 2019-01-07:
when I build welle.io on macOS 10.12 I get a multiple warnings like this:

```
../../../welle.io/src/various/ringbuffer.h:348:13: warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead [-Wdeprecated-declarations]
            PaUtil_FullMemoryBarrier ();
            ^
../../../welle.io/src/various/ringbuffer.h:89:40: note: expanded from macro 'PaUtil_FullMemoryBarrier'
#   define PaUtil_FullMemoryBarrier()  OSMemoryBarrier()
                                       ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has been explicitly marked deprecated here
```

AlbrechtL, 2019-01-08:
We have in mind to replace the ring buffer implementation (https://github.com/AlbrechtL/welle.io/issues/247#issuecomment-440814615).
But this is not a high prior task because the existing implementation is working.

It would be very nice if someone can send a pull request with an alternative implementation.

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL